scientifichackers/ampy #1: Ampy broken

iota-sigma-pi, 2016-09-02:
Unable to access device error on Linux platforms even after disabling debug output

tdicola, 2016-09-03:
You'll want to grab the exact error and paste it here since it will have more details to help troubleshoot, like if it can't open the serial port (which sounds like what could be happening, perhaps you need to run as root with sudo on your distro to access the serial port?) or if it's getting confused by output it didn't expect from the board.  Make sure you're specifying the right port with the --port option or AMPY_PORT environment variable too (run `echo $AMPY_PORT` to see if it's set to anything).  I'll close this for now though since it works on Ubuntu 16.04 for me so it must be something to dig into with your board, serial port, etc.  Good luck!

closed 2016-09-03 by tdicola

benlhy, 2016-09-03:
You are right. I swear I was trying to run this last night and it didn't work. Running:
`sudo ampy --port /dev/ttyUSB0 ls`
yields the expected output. The error was coming from not running with sudo. Earlier I had errors trying to run it with the daily build as well, where using sudo only results in the terminal hanging. I reflashed it with the stable release and it is working.

Tested on Ubuntu 16.04.