scientifichackers/ampy #4: socket.accept fails with ampy.pyboard.PyboardError: timeout waiting for first EOF reception

janakaclk, 2016-09-19:
When I tried to execute https://github.com/micropython/micropython/blob/master/examples/network/http_server_simplistic.py it fails when res = s.accept() is used 

 File "c:\python27\lib\site-packages\ampy\pyboard.py", line 263, in execfile                                                                                                     
    return self.exec_(pyfile)                                                                                                                                                     
  File "c:\python27\lib\site-packages\ampy\pyboard.py", line 255, in exec_                                                                                                        
    ret, ret_err = self.exec_raw(command)                                                                                                                                         
  File "c:\python27\lib\site-packages\ampy\pyboard.py", line 247, in exec_raw                                                                                                     
    return self.follow(timeout, data_consumer)                                                                                                                                    
  File "c:\python27\lib\site-packages\ampy\pyboard.py", line 211, in follow                                                                                                       
    raise PyboardError('timeout waiting for first EOF reception')                                                                                                                 
ampy.pyboard.PyboardError: timeout waiting for first EOF reception 

I have esp.osdebug(None) in the boot.py

Thanks!

tdicola, 2016-09-20:
One thing to help, for future issues can you include the exact ampy command and parameters being run?  That will help speed up troubleshooting issues.

In this case it looks like you're using the ampy run command to run the HTTP server code right?  If so it looks like this is actually by design because you're running code that is waiting for a HTTP connection (the socket accept call).  That wait might be very short if something is instantly trying to connect, or that wait could be very, very long if nothing is waiting to connect.  The way the run command works is it takes your python script and runs it line by line, waiting for each line to finish and running the next.  If a line takes too long (5 seconds IIRC) it will actually timeout and fail with the error you see.

There's a way to make this work though with scripts that run forever or for a long time, add the -n or --no-output option with the run command.  That tells ampy not to wait for output and just run each line one after the other immediately.  You want to use this when your script might run forever or might run for a very long time (greater than 5 seconds).  You won't get any output since ampy doesn't know how long to wait to receive output, but you can actually open the serial REPL with screen or other tools and watch the output from the board.

So to work around this try running with the -n option, like:
```
ampy --port /path/to/serial/port run -n http_server_simplistic.py
```

closed 2016-09-20 by tdicola

janakaclk, 2016-09-25:
Thanks Tony!