scientifichackers/ampy #9: ampy.pyboard.PyboardError: failed to access /dev/ttyUSB0

pengshp, 2016-11-28:
```py
[alarm@archlinux ~]$ ampy -p /dev/ttyUSB0 -b 115200 run -n main.py
Traceback (most recent call last):
  File "/usr/bin/ampy", line 11, in <module>
    sys.exit(cli())
  File "/usr/lib/python3.5/site-packages/click/core.py", line 716, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3.5/site-packages/click/core.py", line 696, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3.5/site-packages/click/core.py", line 1057, in invoke
    Command.invoke(self, ctx)
  File "/usr/lib/python3.5/site-packages/click/core.py", line 889, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3.5/site-packages/click/core.py", line 534, in invoke
    return callback(*args, **kwargs)
  File "/usr/lib/python3.5/site-packages/ampy/cli.py", line 50, in cli
    _board = pyboard.Pyboard(port, baudrate=baud)
  File "/usr/lib/python3.5/site-packages/ampy/pyboard.py", line 143, in __init__
    raise PyboardError('failed to access ' + device)
ampy.pyboard.PyboardError: failed to access /dev/ttyUSB0
```
I can't use it on my nodemcu?what should I do?

tdicola, 2016-12-12:
If this is Linux double check your user has access to the serial port, i.e. try running the screen command to see that the /dev/ttyUSB0 port exists and you can open the REPL to interact with the board:
```
screen /dev/ttyUSB0  115200
```
If that doesn't work then something might be wrong with permissions for the serial port.  Try running with sudo (both screen and the ampy command) to override any permission issue and access as root.

closed 2016-12-12 by tdicola

michelepagot, 2018-02-18:
hi
Happens to me too. Debian 9.3. Ampy 1.0.3. It is not a linux permission issue. screen works fine without to use sudo. More than this...  even ampy ls works at first attempt listing boot.py
Then I try run and it fails. From this point also ls start to fail. Screen  is still functional. Unplugging the board does not help.

VivienGiraud, 2018-03-06:
Same issue here !   
`$ ampy --port /dev/tty.wchusbserial1420 --baud 115200 ls`  

 ```
  Traceback (most recent call last):
  File "/usr/local/bin/ampy", line 11, in <module>
    sys.exit(cli())
  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 1063, in invoke
    Command.invoke(self, ctx)
  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.6/site-packages/ampy/cli.py", line 73, in cli
    _board = pyboard.Pyboard(port, baudrate=baud, rawdelay=delay)
  File "/usr/local/lib/python3.6/site-packages/ampy/pyboard.py", line 147, in __init__
    raise PyboardError('failed to access ' + device)
ampy.pyboard.PyboardError: failed to access /dev/tty.wchusbserial1420

jerryneedell, 2018-04-20:
You cannot have 2 serial connections at the same time. Either Putty(screen, etc) or ampy. Not both.

YashPandit4u, 2018-12-26:
> You cannot have 2 serial connections at the same time. Either Putty(screen, etc) or ampy. Not both.

The exact same issue here. I even terminated all other applications like putty and screen but still showing the same error.

ravindradeoray, 2019-02-01:
I had same issue with my ESP8266 Huzzah board on MacOS. Close terminal with Command + Q and disconnected and connected board. Afterwards I tried to use ampy command and it worked. Hope that helps.

AskforWhat, 2020-02-06:
Maybe it's about file permissions, open the terminal ： 
cd /dev
sudo su
...
chmode 777 ttyUSB0
and exit

lalitkumar09, 2020-06-05:
this issue is because of USB cable we use....  Charging cable that debbug Mobile phone will work fine but not normal cable that do just charging.

LucasCanete, 2020-08-14:
Have the same issue rn with my esp32. I highly suspect this is because i uploaded a program with a while loop to the board and it just runs endlessly so might have to find a way to kill that program first. If somebody knows how to do this in ubuntu i'd be thankful.

V-Perotto, 2022-11-10:
> You cannot have 2 serial connections at the same time. Either Putty(screen, etc) or ampy. Not both.

Thank you very much, that saved me.

I had my board (ESP32) connected by Thonny IDE, when I closed it and tested it in VS Code, I was able to use ampy.

NOTE: Using Windows 10.