scientifichackers/ampy #11: ampy leaves ports open in Windows

carterw, 2016-12-24:
If ampy is unable to complete a serial port IO operation for some reason it will hang. When this happens the user must control-c in order to terminate it and return to a command line prompt. However! The python process keeps running and holds the serial port open. Subsequent invocations of ampy will fail because the port is locked!

The only solution is to either reboot or bring up the Windows Task Manager, look for the python process, and kill it.

tdicola, 2017-01-10:
Ah that's a tricky one since the pyboard.py file from MicroPython's source is what owns and controls the serial connection.  I try not to modify this file since it's pulled from their repo and might change over time, but it does look like I might be able to ensure the close is called on the serial port even if the program crashes or ctrl-c exits.  I just added logic in the latest 1.0.1 version of the tool so it will try to close the serial port no matter what when exiting or crashing.  I couldn't repro it to confirm but try upgrading to the latest ampy and let's see if that helps make ctrl-c not keep the port open.  You can upgrade by running:
```
pip install adafruit-ampy --upgrade
```
Look for the 1.0.1 version to have this change, thanks!

carterw, 2017-01-11:
I installed your version 1.0.1, thanks very much for the update. I'm able to work with the high COM port numbers, no problem. However I am still seeing random situations where ampy hangs for some reason and I have to control-c to exit. It is probably 1 out of 5 times. When this happens I am seeing a zombie python process in the windows task manager and I have to manually kill it. I can then invoke the exact same command and it usually works.

commit b693e22b (2017-01-10): Attempt to fix #11 by adding close of serial on any exit.