scientifichackers/ampy #13: Ampy cannot run all programs

mjmare, 2017-01-04:
I use `ampy run` a lot and find it very useful. However, it turns out that not all programs can be run this way. When the exact same code is run on the board through the REPL (screen) it does work!
I do not understand why. 

The program in question is an example from the micropython-lib umqtt module: example_sub.py

```
import time
from umqtt.simple import MQTTClient

# Publish test messages e.g. with:
# mosquitto_pub -t foo_topic -m hello

# Received messages from subscriptions will be delivered to this callback
def sub_cb(topic, msg):
    print((topic, msg))

def main(server="192.168.2.201"):
    c = MQTTClient("umqtt_client", server)
    c.set_callback(sub_cb)
    print("Before connect")
    c.connect()
    print("After connect")
    c.subscribe(b"foo_topic")
    while True:
        if True:
            # Blocking wait for message
            c.wait_msg()
        else:
            # Non-blocking wait for message
            c.check_msg()
            # Then need to sleep to avoid 100% CPU usage (in a real
            # app other useful actions would be performed instead)
            time.sleep(1)

    c.disconnect()

if __name__ == "__main__":
    main()
```

This is the exception:
```
Traceback (most recent call last):
  File "/Users/mjm/Dropbox/Devel/_Experiments/PyCom/venv3/bin/ampy", line 11, in <module>
    sys.exit(cli())
  File "/Users/mjm/Dropbox/Devel/_Experiments/PyCom/venv3/lib/python3.6/site-packages/click/core.py", line 716, in __call__
    return self.main(*args, **kwargs)
  File "/Users/mjm/Dropbox/Devel/_Experiments/PyCom/venv3/lib/python3.6/site-packages/click/core.py", line 696, in main
    rv = self.invoke(ctx)
  File "/Users/mjm/Dropbox/Devel/_Experiments/PyCom/venv3/lib/python3.6/site-packages/click/core.py", line 1060, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/mjm/Dropbox/Devel/_Experiments/PyCom/venv3/lib/python3.6/site-packages/click/core.py", line 889, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/mjm/Dropbox/Devel/_Experiments/PyCom/venv3/lib/python3.6/site-packages/click/core.py", line 534, in invoke
    return callback(*args, **kwargs)
  File "/Users/mjm/Dropbox/Devel/_Experiments/PyCom/venv3/lib/python3.6/site-packages/ampy/cli.py", line 197, in run
    output = board_files.run(local_file, not no_output)
  File "/Users/mjm/Dropbox/Devel/_Experiments/PyCom/venv3/lib/python3.6/site-packages/ampy/files.py", line 170, in run
    out = self._pyboard.execfile(filename)
  File "/Users/mjm/Dropbox/Devel/_Experiments/PyCom/venv3/lib/python3.6/site-packages/ampy/pyboard.py", line 263, in execfile
    return self.exec_(pyfile)
  File "/Users/mjm/Dropbox/Devel/_Experiments/PyCom/venv3/lib/python3.6/site-packages/ampy/pyboard.py", line 255, in exec_
    ret, ret_err = self.exec_raw(command)
  File "/Users/mjm/Dropbox/Devel/_Experiments/PyCom/venv3/lib/python3.6/site-packages/ampy/pyboard.py", line 247, in exec_raw
    return self.follow(timeout, data_consumer)
  File "/Users/mjm/Dropbox/Devel/_Experiments/PyCom/venv3/lib/python3.6/site-packages/ampy/pyboard.py", line 211, in follow
    raise PyboardError('timeout waiting for first EOF reception')

```

When I put the file on the LoPy board and call it from main.py, it works correctly. This is main.py:

```
from wifi import connect_to_wifi

SSID = 'myssid'
PASSWORD = 'mypassword'
connect_to_wifi(SSID, PASSWORD)

from example_sub_led import main

main()
```

tdicola, 2017-01-09:
Oh if you have a main loop that never exits (the while True) ampy will get confused waiting for the program to finish and eventually timeout.  In those cases add the -n option which tells amp not to wait for output.  For example:
```
ampy run -n foo.py
```

closed 2017-01-09 by tdicola

mjmare, 2017-01-10:
Ah, should have thought of that. Saw you use this feature in one of the videos. OTOH, I would expect amp to "hang", but not the program on the board to not work.

kglasnapp, 2017-04-29:
Thanks -- I was getting timeout error on a microPython program. I used the -n option and it went away. 

The problem is that I do not see the results of any print outs.  How can I avoid the timeout and get printouts.

jurasec, 2017-06-05:
Thank you!, -n works good. Great tool.

EvoltPratom, 2020-05-19:
> Thanks -- I was getting timeout error on a microPython program. I used the -n option and it went away.
> 
> The problem is that I do not see the results of any print outs. How can I avoid the timeout and get printouts.

yes same thing happened try run --help

EvoltPratom, 2020-05-19:
> Thanks -- I was getting timeout error on a microPython program. I used the -n option and it went away.
> 
> The problem is that I do not see the results of any print outs. How can I avoid the timeout and get printouts.

i tried the -n option and intentionally made an error in the code, and ran it, without the -n option it gave a syntax error and with -n it just completed and terminal prompt came back with no error, -n doesn't work