scientifichackers/ampy #15: Add CircuitPython bundle update

tdicola, 2017-01-09:
Consider adding command to download and push the latest bundle of Adafruit CircuitPython drivers to a board.

willingc, 2017-01-10:
Would be a cool addition. Nice work happening on CircuitPython too 💯

carterw, 2017-01-11:
Will a big bundle of drivers fit into RAM? Not much of it is available. I've had to bake portions of my code into the firmware in order to have room to run it.

ladyada, 2019-01-21:
Hiya! We are discontinuing support for ampy, and will no longer be maintaining it. We are leaving this repository available for continued use. If you would like to take over supporting it, please contact us on the Adafruit Discord server and we can transfer the repository to you.
If you wish to continue developing it on your own, please fork the repository.

curiouswala, 2021-02-08:
We have added this feature to the v2 will be doing an alpha release soon.

closed 2021-02-08 by curiouswala