scientifichackers/ampy #17: folder upload

pacmac, 2017-02-20:
Discovered what I think is a bug when updating a folder that already exists.

It appears that if you try and upload a folder that already exists, no changes are made and any new files in the local directory are not copied to the device.

The only way I can see is to overcome this at the moment completely remove the folder on the device and then re-add it again, but this is not very efficient

Cediddi, 2017-09-13:
if we can hash files on both sides, we can overwrite if needed. Bu we need an efficient hashing method for this. Builtin hash function might be good enough but we need to implement the same hash function on ampy to check both files correctly. We might use sha1 too, but sha1 takes our precious ram space and cpu power.

I wonder what adafruit thinks about this approach. I can send a PR.

tdicola, 2017-11-12:
Hrm I can't repro this, could you provide an example of the parameters that are sent to ampy?  Looking at the code too it will just overwrite everything.  Be aware it won't delete any files on the remote board that were removed from the local folder on your computer--it's not a sync and instead more of a push of the current files over to the board.  For a proper sync you want to delete on the remote board then put the folder/files.

closed 2017-11-12 by tdicola

Cediddi, 2017-11-13:
By the way, I opened a PR on this issue but about my comment. Should I open a new issue for 
https://github.com/adafruit/ampy/issues/17#issuecomment-329271799