scientifichackers/ampy #19: ESP8266 Micropython "Could not enter raw repl"

ioncodes, 2017-04-04:
I'm using this board https://www.olimex.com/Products/IoT/MOD-WIFI-ESP8266-DEV/. When I try to run
"sudo ampy --port /dev/ttyUSB0 run test.py" or any other command, it says "Could not enter raw repl". When I enter that, micropython says "raw REPL; CTRL-B to exit".

emilithe, 2017-04-12:
I have similar problem, but with NodeMCU DEVKIT V1.0 board (running MicroPython v1.8.7-7-gb5a1a20a3)

Did you have any code on board running while trying to execute ampy command or any terminal opened? 

In my scenario, ampy (ls command) was not able to enter raw repl while code with **virtual timer initialised** was running on board.
After deinitialising timer (in REPL) and closing terminals ampy works as expected.

g-sam, 2017-05-30:
In my case, the cause was code on the board reading from UART 0.

markserrano915, 2017-06-22:
@g-sam What does it mean that it's reading from UART 0? Does it mean you just need to change something to read on a different UART? If yes, how?

> In my case, the cause was code on the board reading from UART 0.

I had been trying to solve the same problem with my ESP32 for two days now (I think I got the same issue on ESP8266 01 but I probably gave up). But then out of the blue I was able to make it work today.

So when I plugin the board on my Macbook 2013, that's when I get this unable to enter REPL issue. But as soon as I plug it in on my Raspberry Pi, I am able to run the same ampy commands without issues!

My immediate solution is:

1. Do the coding in Mac (because it's more convenient to type in there). 
2. Push to git
3. Pull in the code to my Raspberry Pi
4. Use Raspberry Pi's for uploading code to the board via ampy

By the way I got the same problem with [rshell](https://github.com/dhylands/rshell). I resolved the issue the same way!

Some interesting observations:
* When I use picocom on my Mac, before I am able to enter on the Micropython REPL shell, the terminal spits out some garbage text and afterwards I can work on picocom without any issues (until I use ampy or rshell)
* When I use picocom on my Raspberry, there are no pre-garbage text. I go straight to the REPL with no issues with ampy and rshell

g-sam, 2017-06-22:
I meant I had a sensor plugged into UART 0, which the USB also uses, so there was some kind of conflict. ESP8266 only has one UART-in, so you can't change it. Not sure about the ESP32 and curious if anyone can explain your observations

zjor, 2017-07-07:
I have the same issue on my Mac 10.12.3
`ampy --port /dev/cu.wchusbserial1410 put wget.py`

```
  File "/Library/Python/2.7/site-packages/ampy/pyboard.py", line 184, in enter_raw_repl
    raise PyboardError('could not enter raw repl')
ampy.pyboard.PyboardError: could not enter raw repl
```

sebasarango1180, 2017-07-21:
Hi, has anyone solved this issue? 

I'm trying to connect to ESP32 via serial monitor like this: `screen /dev/ttyUSB0 115200` (which worked just yesterday), but it is not working now. Also, in another terminal, I've used the command `sudo ampy --port /dev/ttyUSB0 put main.py` in order to move files from my computer to the device, but it is not working either.

I use Linux Mint 18, and the USB port is working right.

Thanks for any help.

markserrano915, 2017-07-21:
@sebasarango1180 One thing I discovered is it's important to state who the board maker is. The ESP32 I got from Makerfocus by Geekworm (https://www.amazon.com/dp/B06XXT9ZQZ/ref=cm_sw_r_cp_apa_WGMCzbWRTQC9Z) fails to load Micropython. However it works with Arduino. 

However the one from ESPRESSIF works ( https://www.amazon.com/dp/B01N0SB08Q/ref=cm_sw_r_cp_apa_FIMCzb8HV3CW0) with Micropython. I can list and read files with ampy and rshell.

markserrano915, 2017-07-21:
I've found a workaround for my ESP32 (from Makerfocus by Geekworm).

**Assumptions:**
* You've successfully flashed a Micropython firmware.
* Verify by running **picocom**
* Or verify by running a GUI term like CoolTerm for Mac. 
You should see something like
> MicroPython v1.9.1-218-g56f05137 on 2017-07-01; ESP32 module with ESP32
Type "help()" for more information.
>>>

**The Problem:**
So when I run: 
`ampy --port /dev/tty.wchusbserial1a1210 ls`

I get the following error: "could not enter raw repl":
`File "/usr/local/lib/python2.7/site-packages/ampy/pyboard.py", line 185, in enter_raw_repl
    raise PyboardError('could not enter raw repl')`

**The Workaround:**
1. Open  /usr/local/lib/python2.7/site-packages/ampy/pyboard.py. Find line 171. Specifically go to the **enter_raw_repl** method:
`    def enter_raw_repl(self):
        self.serial.write(b'\r\x03\x03') # ctrl-C twice: interrupt any running program

        # flush input (without relying on serial.flushInput())
        n = self.serial.inWaiting()
        while n > 0:
            self.serial.read(n)
            n = self.serial.inWaiting()`
2.  Right after the **while** loop, add a **time.sleep(2**). So it becomes
`    def enter_raw_repl(self):
        self.serial.write(b'\r\x03\x03') # ctrl-C twice: interrupt any running program

        # flush input (without relying on serial.flushInput())
        n = self.serial.inWaiting()
        while n > 0:
            self.serial.read(n)
            n = self.serial.inWaiting()
        time.sleep(2)`
3. Save and rerun ampy.

You should still be able to run **ampy** with your working boards because we've only added a delay here.

**Other Notes**
1. On a good working board, this is what I see as output (I enabled console logging in the pyboard.py file):
```
$ ampy --port /dev/tty.SLAB_USBtoUART ls

O
r
>
[

boot.py

```
2. On a bad board (Maybe "bad" is a misnomer. It's possible the ampy/pyboard program didn't accommodate special cases from some boards), I see this:
```
mpy --port /dev/tty.wchusbserial1a1210 ls
�
���Ғ�Ҫ�j

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
...
I (2696) cpu_start: Starting scheduler on PRO CPU.
OSError: [Errno 2] ENOENT
MicroPython v1.9.1-218-g56f05137 on 2017-07-01; ESP32 module with ESP32
Type "help()" for more information.
>>>
Traceback (most recent call last):
  File "/usr/local/bin/ampy", line 11, in <module>
    sys.exit(cli())
  ...
  File "/usr/local/lib/python2.7/site-packages/ampy/files.py", line 91, in ls
    self._pyboard.enter_raw_repl()
  File "/usr/local/lib/python2.7/site-packages/ampy/pyboard.py", line 185, in enter_raw_repl
    raise PyboardError('could not enter raw repl')
ampy.pyboard.PyboardError: could not enter raw repl
```

When I run **picocom** on both (ie. picocom /dev/tty.wchusbserial1a1210 -b115200), I can see the Micropython prompt, and I can enter REPL mode on both. So I know REPL mode works on BOTH boards.

The issue is in the ampy/pyboard programs,  and I think it's because of this piece of code:
```
        n = self.serial.inWaiting()
        while n > 0:
            self.serial.read(n)
            n = self.serial.inWaiting()
```

This is essentially like a delay function. The question is should it work the same across all boards?

ioncodes, 2017-07-22:
^ This is how I fixed my issue too!

mcauser, 2017-09-21:
^ adding a `time.sleep(2)` after the `while` in `def enter_raw_repl` fixed ampy on my ESP8266.

philwilkinson40, 2017-10-14:
just to pile on here.  The addition of the time.sleep(1) also worked for NODE MCU ESP8266 .  Has anyone sent this info to Adafruit?

mirhec, 2017-11-01:
Thank you @markserrano915, your solution worked for me as well (Wemos D1 Mini).

tdicola, 2017-11-12:
This is probably a good issue to push up to upstream micropython where the pyboard.py code comes from vs. just here in ampy.  With all the different boards and hardware available it's hard to pick a good sleep that works for one but doesn't slow down others.  Rather than play whack a mole with changing delays to ones which work let's see if upstream micropython might consider a different approach like perhaps a soft reset that skips running main app logic (why the enter raw repl currently does Ctrl-C twice, to stop any running main app).

silps, 2018-02-18:
I had the same issue, I entered with picocom and realised that I had to press twice ctrl-A to enter raw repl. So I modified ampy as following and it started to work for my WEMOS LOLIN32 Lite.

self.serial.write(b'\r\x01') # ctrl-A: enter raw REPL
time.sleep(0.5)
self.serial.write(b'\r\x01') # ctrl-A: enter raw REPL

ptixipivipi, 2018-03-03:
Same issue, same sleep(2) solution working with an ESP32-PICO-KIT V4.

solarkraft, 2018-05-18:
Even a delay of 3 seconds does nothing for me. 
Wemos D1 Mini, Elementary OS 0.4.1 Loki on the first-gen 13" Xiaomi Mi Notebook Air
I am using Python 3.6 via Anaconda, so I edited `/home/paul/miniconda3/lib/python3.6/site-packages/ampy/pyboard.py`. 
```
ampy --port /dev/ttyUSB0 mkdir foo
b'\nR;Ltx'
Traceback (most recent call last):
  File "/home/paul/miniconda3/bin/ampy", line 11, in <module>
    sys.exit(cli())
  File "/home/paul/miniconda3/lib/python3.6/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/paul/miniconda3/lib/python3.6/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/paul/miniconda3/lib/python3.6/site-packages/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/paul/miniconda3/lib/python3.6/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/paul/miniconda3/lib/python3.6/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/paul/miniconda3/lib/python3.6/site-packages/ampy/cli.py", line 127, in mkdir
    exists_okay=exists_okay)
  File "/home/paul/miniconda3/lib/python3.6/site-packages/ampy/files.py", line 120, in mkdir
    self._pyboard.enter_raw_repl()
  File "/home/paul/miniconda3/lib/python3.6/site-packages/ampy/pyboard.py", line 193, in enter_raw_repl
    raise PyboardError('could not enter raw repl')
ampy.pyboard.PyboardError: could not enter raw repl
```
(Takes about 13 seconds)

Running Picocom at the same time gives me outputs like
```
raw EPL CTR-B o eit
>raw REPL CTRL-B o exi
>raw EPL; CTRLB t eit
                     raw EPL; CTRL-B to exit
                                            >ra REPL; CTRL-B to xit
```
I guess this would indicate an unreliable connection, **but** it persists using different cables and text typed through Picocom gets recieved perfectly. 

Any ideas?

silps, 2018-05-20:
@solarkraft may be the code running on the ESP is blocking it somehow or spitting out some stuff.
I would try to reflash it. Meaning to reflash the micropython binary. I something also get frustrated it not copying over the files with ampy and then I just reflash and then usually it works. I also ofter copy the all the rest of the files and then main.py and then the boot.py, as if you copy those the system will try to run the code from them and often for me it was failing to copy any other files after I uploaded boot and main.

luciodj, 2018-05-24:
time.sleep(2) did it for me as well.
Using a Wemos LOLIN32 and Macbook Pro (Sierra): rshell 0.0.12

silps, 2018-05-29:
@solarkraft any luck getting it working? I noticed that the pyboard.py tries to enter raw repl, then makes a soft reboot and then tries to enter raw repl again. It usually works for me, but now I am testing on Linux Windows and Mac OS, it seems to work on Linux and Windows, but on Mac OS the driver seems to reboot the ESP, so it will fail to enter raw repl. I am still investigating more into it. As I am working on an app to update software on the ESP, so I need to make it reliable.

silps, 2018-07-08:
@garthk thanks for sharing. I realized also that the CH304 on MAC OS is reseting my ESP32 all the time when I initiate a serial connection, which sucks and makes ampy not work by default. But adding extra delays makes it work.

ycx315315, 2018-10-22:
i solved my issue. by reinstalling the firmware into the board.

ycx315315, 2018-10-22:
I solved my issue. by reinstalling the firmware into the board.

mrajiullah, 2019-03-19:
Hi!

I get the same problem when I use ampy to work with pycom fipy device. Its pretty unreliable. Any solution related to ampy and fipy? Thanks.

devxpy, 2019-03-19:
@mrajiullah I thought pycom had it's own utils. Do you still need ampy for pycom boards?

Some good news - I've gotten an offer from [Jaaga](https://www.jaaga.in/), who wants to sponsor this project. Hopefully, I can devote more time to solving this issue once we have that sorted out.

CVRELECTRONICA, 2019-09-26:
> He encontrado una solución para mi ESP32 (de Makerfocus by Geekworm).
> 
> **Suposiciones**
> 
> * Has flasheado con éxito un firmware de Micropython.
> * Verifique ejecutando **picocom**
> * O verifique ejecutando un término GUI como CoolTerm para Mac. 
>   Deberías ver algo como
> 
> > MicroPython v1.9.1-218-g56f05137 el 2017-07-01; Módulo ESP32 con ESP32 
> > Escriba "help ()" para obtener más información.
> > > >
> 
> **El problema:**
> cuando corro:
> `ampy --port /dev/tty.wchusbserial1a1210 ls`
> 
> Recibo el siguiente error: "no se pudo ingresar la respuesta cruda":
> `File "/usr/local/lib/python2.7/site-packages/ampy/pyboard.py", line 185, in enter_raw_repl raise PyboardError('could not enter raw repl')`
> 
> **La solución alternativa:**
> 
> 1. Abra /usr/local/lib/python2.7/site-packages/ampy/pyboard.py. Busque la línea 171. Específicamente vaya al método **enter_raw_repl** : 
>    `def enter_raw_repl (self): 
>    self.serial.write (b '\ r \ x03 \ x03') # ctrl-C dos veces: interrumpa cualquier programa en ejecución
>    ```
>     # flush input (without relying on serial.flushInput())
>     n = self.serial.inWaiting()
>     while n > 0:
>         self.serial.read(n)
>         n = self.serial.inWaiting()`
>    ```
> 2. Justo después del **tiempo** de bucle, añadir un **time.sleep (2** ). Entonces se convierte en 
>    'def enter_raw_repl (self): 
>    self.serial.write (b' \ r \ x03 \ x03 ') # ctrl-C dos veces: interrumpe cualquier programa en ejecución
>    ```
>    # flush input (without relying on serial.flushInput())
>    n = self.serial.inWaiting()
>    while n > 0:
>        self.serial.read(n)
>        n = self.serial.inWaiting()
>    time.sleep(2)`
>    ```
> 
> EDITAR: time.sleep (0.5) también funciona, por debajo de 0.5 me sale el temido error de respuesta
> 
> 1. Guarde y vuelva a ejecutar ampy.
> 
> Aún debería poder ejecutar **ampy** con sus tableros de trabajo porque solo hemos agregado un retraso aquí.
> 
> **Otras notas:**
> 
> 1. En una buena placa de trabajo, esto es lo que veo como salida (habilité el registro de la consola en el archivo pyboard.py):
> 
> ```
> $ ampy --port /dev/tty.SLAB_USBtoUART ls
> 
> O
> r
> >
> [
> 
> boot.py
> ```
> 
> 1. En un tablero incorrecto (Quizás "malo" es un nombre inapropiado. Es posible que el programa ampy / pyboard no acomode casos especiales de algunos tableros), veo esto:
> 
> ```
> mpy --port /dev/tty.wchusbserial1a1210 ls
> �
> ���Ғ�Ҫ�j
> 
> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0x00
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> ...
> I (2696) cpu_start: Starting scheduler on PRO CPU.
> OSError: [Errno 2] ENOENT
> MicroPython v1.9.1-218-g56f05137 on 2017-07-01; ESP32 module with ESP32
> Type "help()" for more information.
> >>>
> Traceback (most recent call last):
>   File "/usr/local/bin/ampy", line 11, in <module>
>     sys.exit(cli())
>   ...
>   File "/usr/local/lib/python2.7/site-packages/ampy/files.py", line 91, in ls
>     self._pyboard.enter_raw_repl()
>   File "/usr/local/lib/python2.7/site-packages/ampy/pyboard.py", line 185, in enter_raw_repl
>     raise PyboardError('could not enter raw repl')
> ampy.pyboard.PyboardError: could not enter raw repl
> ```
> 
> Cuando ejecuto **picocom** en ambos (es decir, picocom /dev/tty.wchusbserial1a1210 -b115200), puedo ver el indicador de Micropython y puedo ingresar al modo REPL en ambos. Así que sé que el modo REPL funciona en AMBAS placas.
> 
> El problema está en los programas ampy / pyboard, y creo que es por este código:
> 
> ```
>         n = self.serial.inWaiting()
>         while n > 0:
>             self.serial.read(n)
>             n = self.serial.inWaiting()
> ```
> 
> Esto es esencialmente como una función de retraso. La pregunta es si debería funcionar igual en todos los tableros.
> 
> Editar: creo que mi teoría sobre la función de retraso es incorrecta. Porque en ambas placas, la función de retraso no hace nada. No hace ninguna diferencia. Creo que lo que está sucediendo en mi tablero que no funciona es que escupe texto adicional antes de ir al indicador. Al agregar un time.sleep (), el script solo irá al modo REPL una vez que se haya completado el sueño (esencialmente es cuando se ha escupido el texto adicional. Se supone que el texto adicional solo consume menos que los segundos de tiempo establecidos en el argumento time.sleep ())

Es la solucion perfecta!

neuberfran, 2019-10-18:
Help me pls
https://forum.micropython.org/viewtopic.php?f=18&t=7095

kevinkm, 2021-01-27:
Thanks for markserrano915.
ampy 1.0.7 works on my windows very well, i still have problem on MacOS sierra 10.12.6 through.
I'm able to operate esp32 here but can't connect esp826.  ( I'm just talking about the command ampy)
I add "time.sleep(2)" after "while" in pyboard.py like this:
         while n > 0:
            time.sleep(2)
uninstall the original ampy, then python3 build&install my new ampy, it works now!

AbhijithHaridas, 2021-05-02:
I was using another **REPEL**  instance using **picocom** in the background and I didn't notice it.
I had this same error which leads me to this issue.

The issue went away by closing the other **picocom** instance

ropg, 2021-05-24:
I think I understand it and fixed it, at least for the pyboard.py that lives inside rshell. rshell now has a `-s`, `--suppress-reset` switch.

See https://github.com/dhylands/rshell/issues/27 and the Pull Request linked from there.

wernersbacher, 2022-05-02:
Since adding a delay didn't help it to connect, I just connect via Putty to the COM Port, press Strg+C two times to get into the REPL by hand. Then I delete the main.py with the command `import uos; uos.remove('main.py')` so it doesn't start when I leave the Terminal and then I can copy the files.

jokerElsa, 2023-02-16:
> I solved my issue. by reinstalling the firmware into the board.

me too i'm still working on this error, reinstall the firmware will fix this, but after flashing once, when you want to flash again this error occurs, so everytime i want to modify the code I have to reinstall the firmware which drives me crazy now......

lianrenjujishou, 2023-08-13:
first: os.remove('file running loop')
then: upload file again

sylvinus, 2023-12-21:
I can confirm (like many others?) that the `sleep(2)` suggested above fixes `could not enter raw repl` on the latest mpremote & mpy, on an ESP32 dev board. This should probably be added :)

benfogiel, 2025-08-16:
For anyone still having issues, in my case, disabling soft reset (ie `soft_reset=False`) allows it to enter the raw repl without issue. If using `pyboard.py`, you can do this without changing any code simply by passing the `--no-soft-reset` flag