scientifichackers/ampy #23: Not receiving prints from python application

binloan, 2017-07-19:
I'm running the latest ampy (v1.0.1) and MicroPython v1.9.1-8-g7213e78d from the pre-build binaries with my Feather HUZZAH. 
Everything works fine. I can put and run files with ampy. 
However if a "print()" command is executed, I don't get any output inside ampy. The funny thing though is, I get the error traceback if something bad happened in one of my scripts.

Is there something I missed to set?

tdicola, 2017-11-12:
Be sure you're running ampy with the output capture (i.e. don't run with -n).  Also ampy will only wait about 15 seconds to collect output and then stop and return everything it saw during that time.  If prints happen after that time you won't see them.  You probably want to use a serial terminal program to see the output of your script instead.

closed 2017-11-12 by tdicola

suiluj, 2018-04-06:
@tdicola i tried the serial terminal program like you described [in your YouTube Tutorial at 31:50](https://youtu.be/hrjtAYMrxF4?t=31m50s) . Everything worked great so far but when i use the screen command to see my debug print texts it stops the running program and just opens the repl command prompt.
I use a Wemos D1 Mini connected via usb with [this driver](https://wiki.wemos.cc/downloads)
My port is `/dev/cu.wchusbserial1410` (or  `/dev/tty.wchusbserial1410`). I think because of the driver. Is this the mistake and do i have to install a different driver or am i missing something else?
Do you have a solution or know this problem?
Thank you for your help and great tutorials for micropython.

suiluj, 2018-04-20:
I just flashed a Wemos D1 Mini Pro (16mb flash) and there is no problem with the no output -n flag.

So this seems to be a problem with the driver of the Wemos D1 Mini (4mb flash).
The Wemos D1 Mini uses a different driver and shows as `/dev/tty.wchusbserial1410` 
You can only use the combination `ampy run script.py`(without -n) and afterwards view the print outputs with `screen /dev/tty.wchusbserial1410 115200`. After a while the ampy run command shows the timeout error but the program keeps running. So no problem.

The Wemos D1 Mini Pro (16mb flash) shows as `/dev/tty.SLAB_USBtoUART`
Because of that you are able to use the -n flag:
`ampy run -n script.py` and afterwards view the print outputs with `screen /dev/tty.SLAB_USBtoUART 115200`

**Hint:**
I exported the $AMPY_PORT for the ampy command to keep it short:
`echo "export AMPY_PORT=/dev/tty.SLAB_USBtoUART" >> ~/.bash_profile` (On Mac)
Remember to exit Terminal and open a new Terminal Session so the export takes effect