scientifichackers/ampy #25: Commands only work once on Linux

NyxMC, 2017-08-10:
I already tried this tool on windows and it worked perfectly.
Now, I tried it on Linux (```Linux MINT 18.2 | Kernel 4.10.0-30```) and it's kinda buggy.
If I connect my esp32 running micropython (```esp32-20170810-v1.9.1-397-g3ec00f0d.bin```) and try the command ```ampy -p /dev/ttyUSB0 ls```, I get this output: ```boot.py```. But If I run the command again, nothing happens until I abort using ctrl+c. It starts working again after plugging out/in my ESP32 or pressing the 'EN' button on my board. 

Again, this issue was not present on Windows.
I'd appreciate a fix or a tip what the issue may be so I can fix it myself!

NyxMC, 2017-08-10:
Update 1: It's ```enter_raw_repl()``` in ```pyboard.py``` which blocks...

NyxMC, 2017-08-10:
Update2: It's the 7th line in ```enter_raw_repl()```: ```data = self.read_until(1, b'raw REPL; CTRL-B to exit\r\n>')```.
It seems like the serial port is not properly closed. If I open a serial port to the esp, the command always fails at ```data = self.read_until(1, b'raw REPL; CTRL-B to exit\r\n>')```, but I can see ```raw REPL; CTRL-B to exit``` in the console.

NyxMC, 2017-08-10:
Update 3: This seems to be closely related to #19, but @markserrano915 fix did not work for me.

PawnDrifter, 2017-08-12:
I also see  '`raw REPL; CTRL-B to exit`' in the console at `enter_raw_repl()`:, but no success. #19 fix also did not work for me.
Windows 10 | Virtual box | Ubuntu 16.04. Hardware and USB verified by code execution with Espressif Blink and manual micropython.

You can go to the putty (or similar) window while ampy is waiting and press Ctrl-A, Ctrl-D to step it forward. Then you can see what it was waiting on. It seems that the read_until() function is missing the cues.

I have played around with delays before/after CtrlA and CtrlD, but that isn't a proper fix. It changed the behavior, but predictably didn't solve anything. I'll keep digging as time allows. I'm new to github, ESP32, micropython and all of this, actually. So, I can't propose a solution just yet. Hopefully, I haven't broken protocol here as a complete noob.

ampy is "close, but no cigar" for me so far. I'm not knocking it, just can't figure it out yet.

tdicola, 2017-11-12:
Thanks for raising the issue!  It's hard to act on this though--I haven't seen the same issue in Linux (Ubuntu) and can't investigate.  Let's close the issue for now but folks can chime in or re-open if more reports occur.  Thanks!

closed 2017-11-12 by tdicola

beary, 2018-09-26:
Meet the same problem. I have to pull out my esp32 after any `ampy` commands.

- ubuntu 18.04
- picocom v2.2

thejoeejoee, 2019-04-29:
same problem, first is OK, second fails

```
ampy -p /dev/ttyUSB0 put config.json
ampy -p /dev/ttyUSB0 put _install.py
b'\x03\x03 \x07\xe7;\xe4'
Traceback (most recent call last):
  File "/home/thejoeejoee/.local/bin/ampy", line 11, in <module>
    sys.exit(cli())
  File "/home/thejoeejoee/.local/lib/python3.6/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/thejoeejoee/.local/lib/python3.6/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/thejoeejoee/.local/lib/python3.6/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/thejoeejoee/.local/lib/python3.6/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/thejoeejoee/.local/lib/python3.6/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/thejoeejoee/.local/lib/python3.6/site-packages/ampy/cli.py", line 263, in put
    board_files.put(remote, infile.read())
  File "/home/thejoeejoee/.local/lib/python3.6/site-packages/ampy/files.py", line 208, in put
    self._pyboard.enter_raw_repl()
  File "/home/thejoeejoee/.local/lib/python3.6/site-packages/ampy/pyboard.py", line 192, in enter_raw_repl
    raise PyboardError('could not enter raw repl')
ampy.pyboard.PyboardError: could not enter raw repl
```

versions
- micropython 1.10
- ampy 1.0.7
- ubuntu 18.10

bulletmark, 2021-03-23:
For me, the first two `ampy` commands work (after a board reset) then the 3rd and any subsequent always fail(!) with the `could not enter raw repl` error, or it just hangs. Exact same thing happens using `pyboard.py` (latest version) from MicroPython.

AbhijithHaridas, 2021-05-02:
I was using another **REPEL**  instance using **picocom** in the background and I didn't notice it.
I had this same error which leads me to this issue.

The issue went away by closing the other **picocom** instance