scientifichackers/ampy #32: KeyboardInterrupt on Windows

dobra-dobra, 2017-10-26:
Something seems not working with rising a KeyboardInterrupt on Windows 7, at least in my case. In command line window Ctrl+C does not stop program execution (I tried this combination also in ipython, where it works as expected, so it's not keyboard or terminal problem). I also tried to use Ctrl+Break combination, which closes process (new line with prompt appears), but program is still running on the board (I have a program which turns on/off relay every second, so I can hear a click when it is running).

larsks, 2017-11-11:
@dobra-dobra Are you reporting a problem with ampy, or with CircuitPython?

tdicola, 2017-11-12:
This doesn't appear to be related to ampy so closing it, thanks!

closed 2017-11-12 by tdicola