scientifichackers/ampy #35: print filesizes with ls command

kanalizator, 2017-12-02:
Hello, thanks for this wonderful utility.
Only thing I am missing is file size info for ls command - it would be the easiest check if the file is same as on my PC.
So i suggest adding something like -l parameter to show more about files, than just names.

Thanks.

ladyada, 2018-02-08:
@tdicola next tidyday, implement this with an -l parameter as suggested (to mimic "ls -l")

tdicola, 2018-02-20:
This is implemented now in the latest master branch code.  Note that MicroPython doesn't calculate the size of directories and there's no way to detect a file from directory so they'll show as 0 bytes.
```
tony-imac:mpy tony$ ampy ls -l
test.txt - 34 bytes
test2.txt - 35 bytes
test - 0 bytes
.fseventsd - 0 bytes
lib - 0 bytes
Windows 7 Driver - 0 bytes
.metadata_never_index - 0 bytes
.Trashes - 0 bytes
boot_out.txt - 80 bytes
README.txt - 2103 bytes
```

closed 2018-02-20 by tdicola