scientifichackers/ampy #36: Display free space on storage medium

microbuilder, 2017-12-29:
In addition to the comment in #35, which seems like a good idea, it might be useful to know how much free space is left or how much space has been used (which might be easier to calculate?) in flash memory?

ladyada, 2018-02-08:
@tdicola next tidyday, implement this along with https://github.com/adafruit/ampy/issues/35 
eg.  print out total/free space on the last line

tdicola, 2018-02-20:
Unfortunately this doesn't appear to be possible.  There's no command in the os or uos module to get global filesystem info.  Attempting to call stat on the root ('/' or '') returns 0 total size, and attempting to call the statvfs command also returns no usable info (0 size).

closed 2018-02-20 by tdicola