scientifichackers/ampy #39: Problem with ESP32

zpowellman, 2018-01-15:
ampy works well using wemos d1 mini, but has some problems with the ESP_Core_Board_V2
If I issue "ampy ls" command, the files in the directory appear.  Issuing same command again, ampy hangs.

"ampy put file.py" sends file OK, but ampy never returns.  Tried on 2 different modules.

ds2k5, 2018-01-28:
ampy 1.2.6 with ESP-WROOM-32

**ampy --port /dev/ttyUSB0 ls**

hang no respond

**lsusb**

Bus 001 Device 004: ID 10c4:ea60 Cygnal Integrated Products, Inc. CP210x UART Bridge / myAVR mySmartUSB light

esptool v2.2.1

zpowellman, 2018-01-28:
[ampyesp32.txt](https://github.com/adafruit/ampy/files/1671433/ampyesp32.txt)
[image]
Here's my experience. jpg was taken after programming and reset.  puTTY was then closed and ampy command issued.  Board is ESP core_board_V2.   
Thanks for looking
Best Regards,
Dan Powell

closed 2018-01-28 by ladyada