scientifichackers/ampy #40: No file/directory found

vijaykarthik123, 2018-02-02:
[image]

Is this a bug? Am I doing something wrong with my run command?

evemorgen, 2018-02-05:
It might seems a bit counterintuitive but, when you type `run scratch.py` ampy looks for `scratch.py` file in local machine current directory ("C:\Python27\Scripts" in this case), not board filesystem. However, it would be cool tho to actually run things from board filesystem. Correct me if I'm wrong.

vijaykarthik123, 2018-02-05:
I couldn't get it to work this way and hence I've resorted to importing `scratch.py` to` main.py` and then using the variables there. 

But to your point, if I replace my main.py with `scratch.py` and run it using the command, `ampy -p com4 run main.py` it works.

tdicola, 2018-02-07:
Yeah run works by copying the local file to the board and running it, not by running a file already on the board.  If you're curious check the help output to see a good description of what it does (and other commands):
```
Run a script and print its output.
    Run will send the specified file to the board and execute it immediately.
    Any output from the board will be printed to the console (note that this is
    not a 'shell' and you can't send input to the program).
    Note that if your code has a main or infinite loop you should add the --no-output
    option.  This will run the script and immediately exit without waiting for
    the script to finish and print output.
    For example to run a test.py script and print any output after it finishes:
      ampy --port /board/serial/port run test.py
    Or to run test.py and not wait for it to finish:
      ampy --port /board/serial/port run --no-output test.py
```
If you missed it too there's a good guide that describes the usage of the commands here: https://learn.adafruit.com/micropython-basics-load-files-and-run-code/run-code

vijaykarthik123, 2018-02-07:
Okay, thanks, mate! I have currently moved on from Windows CMD prompt to Ubuntu, which is a lot easier. I'll keep the above things in mind, though. GD^.

closed 2018-02-20 by tdicola