scientifichackers/ampy #41: Can not run twice for ESP32

iascchen, 2018-03-02:
When I use it on my MAC OS, I found it will be hanged when i run `ampy --port /dev/tty.SLAB_USBtoUART ls` in secend time. 

I used the ampy 1.0.3 version, and micropython firmware of esp32-20180302-v1.9.3-386-g9884a2c7.bin, python version is 3.6.4

`
$ ampy --port /dev/tty.SLAB_USBtoUART ls
boot.py
lesson_2_led.py
main.py
$ ampy --port /dev/tty.SLAB_USBtoUART ls

b'\x03\x02`\x0f\xeer\xec'
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.6/bin/ampy", line 11, in <module>
    sys.exit(cli())
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/ampy/cli.py", line 147, in ls
    for f in board_files.ls(directory):
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/ampy/files.py", line 94, in ls
    self._pyboard.enter_raw_repl()
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/ampy/pyboard.py", line 192, in enter_raw_repl
    raise PyboardError('could not enter raw repl')
ampy.pyboard.PyboardError: could not enter raw repl
`

I checked the source code of micropython, and think may be the issue caused by it firmware code in ports/esp32/main.c

`
    for (;;) {
        if (pyexec_mode_kind == PYEXEC_MODE_RAW_REPL) {
            if (pyexec_raw_repl() != 0) {
                break;
            }
        } else {
            if (pyexec_friendly_repl() != 0) {
                break;
            }
        }
    }
`

hutualive, 2018-05-14:
I have similar issue on mac but not linux(debian). I checked the code mentioned in ports/esp32/main.c, seems it's a standard implementation cross different Arch/MCUs, so better figure out a walk-around on pyboard.py alone.

hyansuper, 2018-10-20:
I have the same issue, I am using Ubuntu

Tomekske, 2019-01-05:
Use the option '-d 0.5' in every command.
e.g.: ampy -p COM5 -d 0.5 ls or ampy -p COM5 -d 0.5 put file.py

ladyada, 2019-01-21:
Hiya! We are discontinuing support for ampy, and will no longer be maintaining it. We are leaving this repository available for continued use. If you would like to take over supporting it, please contact us on the Adafruit Discord server and we can transfer the repository to you.
If you wish to continue developing it on your own, please fork the repository.

closed 2019-01-21 by ladyada

scientist1642, 2019-02-16:
or if `-d 0.5` doesn't work use `-d 1` from Mac.