scientifichackers/ampy #43: "could not enter raw repl"

larsks, 2018-04-05:
When running `ampy` several times in succession, I often see fails that look like this:

```
b'PL; CTRL-B to exit\r\n>'
Traceback (most recent call last):
  File "/home/lars/.local/bin/ampy", line 11, in <module>
    load_entry_point('adafruit-ampy', 'console_scripts', 'ampy')()
  File "/home/lars/.local/lib/python3.6/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/lars/.local/lib/python3.6/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/lars/.local/lib/python3.6/site-packages/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/lars/.local/lib/python3.6/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/lars/.local/lib/python3.6/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/lars/src/ampy/ampy/cli.py", line 220, in put
    board_files.put(remote, infile.read())
  File "/home/lars/src/ampy/ampy/files.py", line 157, in put
    self._pyboard.enter_raw_repl()
  File "/home/lars/src/ampy/ampy/pyboard.py", line 192, in enter_raw_repl
    raise PyboardError('could not enter raw repl')
ampy.pyboard.PyboardError: could not enter raw repl
```

The issue is that it obviously *was* able to enter raw repl mode; that's why we see `b'PL; CTRL-B to exit\r\n>'` printed at the top there (that comes from https://github.com/adafruit/ampy/blob/ff35f29b5b668f2555e52cf12a0d320f93f93c16/ampy/pyboard.py#L191).

trailhead, 2018-04-07:
This solution worked for me.
https://github.com/adafruit/ampy/issues/19

larsks, 2018-04-07:
@trailhead thanks for the pointer; I've actually implemented a different solution locally. Since `ampy` was obviously reading the REPL prompt (just not the whole thing), I modified the sentinel condition so that it looks like:

```
diff --git a/ampy/pyboard.py b/ampy/pyboard.py
index fa503d4..a21ec1a 100644
--- a/ampy/pyboard.py
+++ b/ampy/pyboard.py
@@ -187,7 +187,7 @@ class Pyboard:
 
         self.serial.write(b'\r\x01') # ctrl-A: enter raw REPL
         data = self.read_until(1, b'raw REPL; CTRL-B to exit\r\n>')
-        if not data.endswith(b'raw REPL; CTRL-B to exit\r\n>'):
+        if not data.endswith(b'CTRL-B to exit\r\n>'):
             print(data)
             raise PyboardError('could not enter raw repl')
```

And now it works reliably without requiring a `sleep`.

ladyada, 2019-01-21:
Hiya! We are discontinuing support for ampy, and will no longer be maintaining it. We are leaving this repository available for continued use. If you would like to take over supporting it, please contact us on the Adafruit Discord server and we can transfer the repository to you.
If you wish to continue developing it on your own, please fork the repository.

devxpy, 2019-02-27:
Please continue the discussion over at https://github.com/pycampers/ampy/issues/19

closed 2019-02-27 by devxpy

datajett, 2020-01-05:
> @trailhead thanks for the pointer; I've actually implemented a different solution locally. Since `ampy` was obviously reading the REPL prompt (just not the whole thing), I modified the sentinel [...]

this one nailed it for me, after lookig for days & seeing many solutions that didnt work. 
great job.