scientifichackers/ampy #44: Cannot distinguish directories from files with ls command

joewez, 2018-04-21:
One nice enhancement would be the ability to distinguish files from directories with the ls (or some other) command.  Currently I distinguish between the two by the presence or absence of an extension but I am not sure that will always work.

ladyada, 2019-01-21:
Hiya! We are discontinuing support for ampy, and will no longer be maintaining it. We are leaving this repository available for continued use. If you would like to take over supporting it, please contact us on the Adafruit Discord server and we can transfer the repository to you.
If you wish to continue developing it on your own, please fork the repository.

devxpy, 2019-01-22:
I see that this was considered earlier - https://github.com/pycampers/ampy/pull/18.

Although, this could probably be added to the newer `ls -l`, so I'll keep this open.

devxpy, 2019-02-25:
This is added in the [dev](https://github.com/pycampers/ampy/tree/dev) branch, it will be released later.

closed 2019-02-25 by devxpy