scientifichackers/ampy #45: Ampy able to upload binary files... but unable to download them.

joewez, 2018-04-27:
Not a major issue, but I was just wondering if this was more of an oversight rather than an actual deficiency.  I'm mean if it works one way... why not the other? 

I am running Python 3 on Windows 10.

ladyada, 2019-01-21:
Hiya! We are discontinuing support for ampy, and will no longer be maintaining it. We are leaving this repository available for continued use. If you would like to take over supporting it, please contact us on the Adafruit Discord server and we can transfer the repository to you.
If you wish to continue developing it on your own, please fork the repository.

curiouswala, 2021-02-08:
Currently the file transfer happens over Serial communication which have a lot of bugs specially in the case of binary files. We will be releasing a v2 soon which works over WIFI and solves the above problem.

closed 2021-02-08 by curiouswala