scientifichackers/ampy #47: GET command does not work for feather52

jerryneedell, 2018-05-07:
The get command does not work when accessing my nrf52 - feather52 board via ampy.

ls and put work, but no get

Note: this example includes PR #46 but the issues was also present without it.

```
jerryneedell@Ubuntu-Macmini:~/projects/feather52$ ampy -d 1.5 -p /dev/ttyUSB0  put test.py
jerryneedell@Ubuntu-Macmini:~/projects/feather52$ ampy -d 1.5 -p /dev/ttyUSB0  ls
test.py
my_scan.py
jerryneedell@Ubuntu-Macmini:~/projects/feather52$ ampy -d 1.5 -p /dev/ttyUSB0  get test.py
Traceback (most recent call last):
  File "/usr/local/bin/ampy", line 11, in <module>
    load_entry_point('adafruit-ampy==1.0.4', 'console_scripts', 'ampy')()
  File "/usr/local/lib/python3.5/dist-packages/click-6.7-py3.5.egg/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/click-6.7-py3.5.egg/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.5/dist-packages/click-6.7-py3.5.egg/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.5/dist-packages/click-6.7-py3.5.egg/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.5/dist-packages/click-6.7-py3.5.egg/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/adafruit_ampy-1.0.4-py3.5.egg/ampy/cli.py", line 99, in get
  File "/usr/local/lib/python3.5/dist-packages/adafruit_ampy-1.0.4-py3.5.egg/ampy/files.py", line 77, in get
  File "/usr/local/lib/python3.5/dist-packages/adafruit_ampy-1.0.4-py3.5.egg/ampy/files.py", line 70, in get
  File "/usr/local/lib/python3.5/dist-packages/adafruit_ampy-1.0.4-py3.5.egg/ampy/pyboard.py", line 267, in exec_
ampy.pyboard.PyboardError: ('exception', b'', b'Traceback (most recent call last):\r\n  File "<stdin>", line 8, in <module>\r\n  File "<stdin>", line 8, in <module>\r\nAttributeError: \'module\' object has no attribute \'stdout\'\r\n')
```

jerryneedell, 2018-05-07:
I guess the error message says it all:
https://github.com/adafruit/ampy/blob/master/ampy/files.py#L66
ties to use sys.stdout.write  which does not exist on the nrf52 build

feather52
```
Adafruit CircuitPython 3.0.0-alpha.6-46-gd42c83f-dirty on 2018-05-07; Bluefruit nRF52 Feather with NRF52832
>>> import sys
>>> dir(sys)
['__name__', 'path', 'argv', 'version', 'version_info', 'implementation', 'platform', 'byteorder', 'maxsize', 'exit', 'modules', 'print_exception']
>>> 
```

Note: it works for the ESP8266 which has support for sys.stdout
esp8266
```
Adafruit CircuitPython 3.0.0-alpha.6-46-gd42c83f-dirty on 2018-05-07; ESP module with ESP8266
>>> import sys
>>> dir(sys)
['__name__', 'path', 'argv', 'version', 'version_info', 'implementation', 'platform', 'byteorder', 'maxsize', 'exit', 'stdin', 'stdout', 'stderr', 'modules', 'print_exception']
>>> dir(sys.stdout)
['buffer', 'read', 'readinto', 'readline', 'readlines', 'write', 'close', '__del__', '__enter__', '__exit__']
>>> 

```

ladyada, 2019-01-21:
Hiya! We are discontinuing support for ampy, and will no longer be maintaining it. We are leaving this repository available for continued use. If you would like to take over supporting it, please contact us on the Adafruit Discord server and we can transfer the repository to you.
If you wish to continue developing it on your own, please fork the repository.

devxpy, 2019-01-22:
Huh, that looks weird, shouldn't there just be a `print` call instead? Can you try replacing `sys.stdout.write` with just `print`?