scientifichackers/ampy #48: GET, PUT and LS commands do not work with esp8266

RobertoDebarba, 2018-06-08:
GET, PUT and LS commands do not work with esp8266 via ampy.

Firmware: Micropython 1.9.4
Command: `sudo ampy --port /dev/ttyUSB0 --baud 115200 ls`

```
Traceback (most recent call last):
  File "/usr/local/bin/ampy", line 11, in <module>
    load_entry_point('adafruit-ampy==1.0.3', 'console_scripts', 'ampy')()
  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/ampy/cli.py", line 147, in ls
    for f in board_files.ls(directory):
  File "/usr/local/lib/python3.6/dist-packages/ampy/files.py", line 103, in ls
    raise ex
  File "/usr/local/lib/python3.6/dist-packages/ampy/files.py", line 96, in ls
    out = self._pyboard.exec_(textwrap.dedent(command))
  File "/usr/local/lib/python3.6/dist-packages/ampy/pyboard.py", line 265, in exec_
    raise PyboardError('exception', ret, ret_err)
ampy.pyboard.PyboardError: ('exception', b'', b'Traceback (most recent call last):\r\n  File "<stdin>", line 6, in <module>\r\nAttributeError: no such attribute\r\n')
```

MikeTeachman, 2018-06-13:
For me, it works running Ampy on Windows 7, Adafruit ESP8266 Huzzah.  Here is what I see with v1.9.4:

C:\test>ampy --version
ampy, version 1.0.3
C:\test>ampy -pCOM10 -d1 ls
boot.py
main.py
C:\test>ampy -pCOM10 -d1 get boot.py my_boot.py

Adding the "-d1" delay is critical for getting Ampy to work with my ESP8266 devices.

RobertoDebarba, 2018-06-14:
I downloaded micropython 1.9.4 again, erase flash and write. And now ... It works!
I dont now if I did something wrong on first time, but tried everything e nothing worked.

Anyway, thanks for your support!

closed 2018-06-14 by RobertoDebarba