scientifichackers/ampy #49: Ampy is install yet linux indicates command not found.

qwerty12345asdf, 2018-06-15:
I am trying to access my nodemcu file system using the ampy tool by trying to follow this guide.
I have run into the following problem even after installing ampy the system simply wont allow me to use it says ampy not installed.
Hope you can help me I have attached a screen-shot of the method i used to install and verify if ampy was installed or not.
[image]

qwerty12345asdf, 2018-06-15:
The guide i used was https://cdn-learn.adafruit.com/downloads/pdf/micropython-basics-load-files-and-run-code.pdf

qwerty12345asdf, 2018-06-15:
used git sorry for the trouble :)

closed 2018-06-15 by qwerty12345asdf

IO42630, 2018-12-03:
Thank you for the hint. This fixed the issue for me:

`pip3 install --upgrade git+https://github.com/adafruit/ampy`

The `README.md`  is a bit misleading.