scientifichackers/ampy #50: Cannot rm a file or mkdir on ESP32

jusmop, 2018-06-23:
I have installed MicroPython 1.9.4 on a ESP32-Tiny with chip ESP32D0WDQ6 revision 1

I can connect to the board fine over the USB/serial interface, get an REPL prompt, connect to wifi, drive I/O, and upload files. All good.

But I cannot delete a file or create a directory.

E.g.:
$ ampy --port /dev/ttyUSB0 mkdir test
Traceback (most recent call last):
  File "/usr/local/bin/ampy", line 11, in <module>
    load_entry_point('adafruit-ampy==1.0.3', 'console_scripts', 'ampy')()
  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/ampy/cli.py", line 127, in mkdir
    exists_okay=exists_okay)
  File "/usr/local/lib/python3.6/dist-packages/ampy/files.py", line 129, in mkdir
    raise ex
  File "/usr/local/lib/python3.6/dist-packages/ampy/files.py", line 122, in mkdir
    out = self._pyboard.exec_(textwrap.dedent(command))
  File "/usr/local/lib/python3.6/dist-packages/ampy/pyboard.py", line 265, in exec_
    raise PyboardError('exception', ret, ret_err)
ampy.pyboard.PyboardError: ('exception', b'', b'Traceback (most recent call last):\r\n  File "<stdin>", line 6, in <module>\r\n  File "/lib/os/__init__.py", line 80, in mkdir\r\nNameError: name \'mkdir_\' is not defined\r\n')

It's like a library is not installed. Or there is something funny going on with those "\r\n"s... My host box is running linux.

Any ideas what is wrong?

Thanks,
Justin.

dybber, 2018-08-02:
Do you still have this problem? I can make directories and remove files just fine. Sparkfun ESP32 Thing with the "esp32-20180511-v1.9.4.bin" firmware

jusmop, 2018-08-02:
I no longer have this issue - I re-flashed the firmware and it worked. I
should have updated the issue. It was odd though.

closed 2018-08-02 by ladyada