scientifichackers/ampy #55: ls does not work for subdirectories

jerryneedell, 2018-07-20:
This issue has been present for awhile but I forgot to file an isse on it -- here it is failing on an ESP8266, but the same problem exists for the nrf52

any attempt to use the "ls" command on a sub-directory fails and it leaves the board in the "Raw REPL"
A control-B brings it back to normal REPL

I am using ampy 1.0.5 with the following .ampy file
```
jerryneedell@Ubuntu-Macmini:~/projects/esp8266/ampy$ cat .ampy
# Example .ampy file
# Please fill in your own port, baud rate, and delay
AMPY_PORT=/dev/ttyUSB0
AMPY_BAUD=115200
AMPY_DELAY=0

```
create subdirectory lib and copy a file to the lib subdirectory
```
jerryneedell@Ubuntu-Macmini:~/projects/esp8266/ampy$ ampy mkdir lib
jerryneedell@Ubuntu-Macmini:~/projects/esp8266/ampy$ ampy put ~/projects/Adafruit_Circuitpython_Bundle/3.x/lib/neopixel.mpy lib/neopixel.mpy
jerryneedell@Ubuntu-Macmini:~/projects/esp8266/ampy$ ampy put jewel.py
jerryneedell@Ubuntu-Macmini:~/projects/esp8266/ampy$ ampy ls
boot.py
lib
jewel.py
```
using os.listdir, - verify that the files are present

```Adafruit CircuitPython 3.0.0-rc.0-157-g1fb8197 on 2018-07-19; ESP module with ESP8266
>>> 
>>> 
>>> import os
>>> os.listdir()
['boot.py', 'lib', 'jewel.py']
>>> os.listdir('lib')
['neopixel.mpy']
>>> os.listdir('lib')

```

Here are various attempts to us "ls"
```
jerryneedell@Ubuntu-Macmini:~/projects/esp8266/ampy$ ampy ls lib
Traceback (most recent call last):
  File "/usr/local/bin/ampy", line 11, in <module>
    load_entry_point('adafruit-ampy==1.0.5', 'console_scripts', 'ampy')()
  File "/usr/local/lib/python3.5/dist-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.5/dist-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.5/dist-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.5/dist-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/ampy/cli.py", line 162, in ls
    for f in board_files.ls(directory, long_format=long_format):
  File "/usr/local/lib/python3.5/dist-packages/ampy/files.py", line 124, in ls
    raise ex
  File "/usr/local/lib/python3.5/dist-packages/ampy/files.py", line 117, in ls
    out = self._pyboard.exec_(textwrap.dedent(command))
  File "/usr/local/lib/python3.5/dist-packages/ampy/pyboard.py", line 267, in exec_
    raise PyboardError('exception', ret, ret_err)
ampy.pyboard.PyboardError: ('exception', b'', b'Traceback (most recent call last):\r\n  File "<stdin>", line 6, in <module>\r\nOSError: [Errno 22] Invalid argument\r\n')
```
```
jerryneedell@Ubuntu-Macmini:~/projects/esp8266/ampy$ ampy ls lib/
Traceback (most recent call last):
  File "/usr/local/bin/ampy", line 11, in <module>
    load_entry_point('adafruit-ampy==1.0.5', 'console_scripts', 'ampy')()
  File "/usr/local/lib/python3.5/dist-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.5/dist-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.5/dist-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.5/dist-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/ampy/cli.py", line 162, in ls
    for f in board_files.ls(directory, long_format=long_format):
  File "/usr/local/lib/python3.5/dist-packages/ampy/files.py", line 124, in ls
    raise ex
  File "/usr/local/lib/python3.5/dist-packages/ampy/files.py", line 117, in ls
    out = self._pyboard.exec_(textwrap.dedent(command))
  File "/usr/local/lib/python3.5/dist-packages/ampy/pyboard.py", line 267, in exec_
    raise PyboardError('exception', ret, ret_err)
ampy.pyboard.PyboardError: ('exception', b'', b'Traceback (most recent call last):\r\n  File "<stdin>", line 6, in <module>\r\nOSError: [Errno 22] Invalid argument\r\n')
```
```
jerryneedell@Ubuntu-Macmini:~/projects/esp8266/ampy$ ampy ls /lib
Traceback (most recent call last):
  File "/usr/local/bin/ampy", line 11, in <module>
    load_entry_point('adafruit-ampy==1.0.5', 'console_scripts', 'ampy')()
  File "/usr/local/lib/python3.5/dist-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.5/dist-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.5/dist-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.5/dist-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/ampy/cli.py", line 162, in ls
    for f in board_files.ls(directory, long_format=long_format):
  File "/usr/local/lib/python3.5/dist-packages/ampy/files.py", line 124, in ls
    raise ex
  File "/usr/local/lib/python3.5/dist-packages/ampy/files.py", line 117, in ls
    out = self._pyboard.exec_(textwrap.dedent(command))
  File "/usr/local/lib/python3.5/dist-packages/ampy/pyboard.py", line 267, in exec_
    raise PyboardError('exception', ret, ret_err)
ampy.pyboard.PyboardError: ('exception', b'', b'Traceback (most recent call last):\r\n  File "<stdin>", line 6, in <module>\r\nOSError: [Errno 22] Invalid argument\r\n')
```

jerryneedell, 2018-07-20:
FYI -- here is the error on a feather 52832
```
jerryneedell@Ubuntu-Macmini:~/projects/feather52$ ampy ls
lib
sdmount_lib.py
ble_scan.py
sdmount.py
stmpe610_i2c.py
adafruit_stmpe610.mpy
i2c_scan.py
stmpe610_spi.py
paint_nrf52.py
am2320_test.py
I2S_Test_Script.py
stmpe.py
jerryneedell@Ubuntu-Macmini:~/projects/feather52$ ampy ls lib
Traceback (most recent call last):
  File "/usr/local/bin/ampy", line 11, in <module>
    load_entry_point('adafruit-ampy==1.0.5', 'console_scripts', 'ampy')()
  File "/usr/local/lib/python3.5/dist-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.5/dist-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.5/dist-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.5/dist-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/ampy/cli.py", line 162, in ls
    for f in board_files.ls(directory, long_format=long_format):
  File "/usr/local/lib/python3.5/dist-packages/ampy/files.py", line 124, in ls
    raise ex
  File "/usr/local/lib/python3.5/dist-packages/ampy/files.py", line 117, in ls
    out = self._pyboard.exec_(textwrap.dedent(command))
  File "/usr/local/lib/python3.5/dist-packages/ampy/pyboard.py", line 267, in exec_
    raise PyboardError('exception', ret, ret_err)
ampy.pyboard.PyboardError: ('exception', b'', b'Traceback (most recent call last):\r\n  File "<stdin>", line 6, in <module>\r\nOSError: [Errno 22] EINVAL\r\n')
```

devxpy, 2018-07-20:
Can you test my code ^^^ on your feather?

jerryneedell, 2018-07-20:
tested on esp8266 feather and on feather52832 (nrf52)  -- see notes in the PR #56 
Thank you!

PR #56 (2018-07-23): fixed bugs with 'ls'. Added recursive to 'ls', along with some beauty…
- fix #55.
- Add a recursive mode to `ls` command.
- Always display the file/dir names as the full path with a leading slash for least confusion.
- Code format with black (my editor was rigged to do it, sorry).

*Tested on Wemos D1 Mini*

----

Here are some examples of what I did -

(`...` == ` -p /dev/ttyUSB0 -b 115200`)

```
⋊ ~/ampy> ampy ... ls -r -l
/boot.py - 0 bytes
/glove/common/__init__.mpy - 82 bytes
/glove/common/config.mpy - 302 bytes
/glove/common/unetwork.mpy - 5956 bytes
/glove/micropython/__init__.mpy - 82  [... cut]
closed 2018-07-23 by tannewt