scientifichackers/ampy #59: Changes to ls are great, but I need version 1.0.3 back

joewez, 2018-08-02:
I have a downstream application that utilizes the output of the ls command to function.

     https://github.com/joewez/AmpyFileManager

Unfortunately, the latest changes to the output formats has totally thrashed my app!  

While I mean to update the application to work with the new formatting ASAP, I have a couple of non-specific requests for help...

1) How do I extract version 1.0.3 from the source on github?  I know I should have it as soon as I clone the repo, but I'm not git savvy enough to switch to before the latest changes.

2) Could you consider a means of differentiating the files from the sub-directories with the ls command?  As it is, the only general difference is the lack of an extension.

Anyway, thanks everyone for their great work!

dhalbert, 2018-08-02:
I'll add tags that mark the versions. Then you can do `git checkout 1.0.3`. I'll comment here when that's done.

Please write up 2. as a separate issue, maybe with some examples.

joewez, 2018-08-02:
Thanks for marking the versions.
I already have a ticket for 2.  I will see if I can flesh it out with some examples.

dhalbert, 2018-08-02:
OK, made lots of tags for lots of previous versions. The initial version was 0.9.0, and then it went back to 0.5.0, so I skipped the initial version, since it was out of order.

Do `git pull --tags` if you cloned the original, or `git fetch --tags adafruit` if you're using a remote called `adafruit` (or `upstream` for `upstream`, etc.).

devxpy, 2019-02-25:
Hey! Sorry, you had this experience.

I plan to add a more computer-readable output (preferably JSON) for other tools to parse. 

Your GUI project looks great, looking forward to having first-class support for stuff like this.