scientifichackers/ampy #60: Latest changes to ls failing in Windows 10 for sub-directories

joewez, 2018-08-02:
Upgrading to 1.0.5...

I can no longer list files in a subdirectory...

`
D:\Development>ampy -p com4 -b 115200 ls /wwwroot/
Traceback (most recent call last):
  File "C:\Users\Joseph\AppData\Local\Programs\Python\Python36-32\Scripts\ampy-script.py", line 11, in <module>
    load_entry_point('adafruit-ampy==1.0.5', 'console_scripts', 'ampy')()
  File "c:\users\joseph\appdata\local\programs\python\python36-32\lib\site-packages\click\core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\joseph\appdata\local\programs\python\python36-32\lib\site-packages\click\core.py", line 697, in main
    rv = self.invoke(ctx)
  File "c:\users\joseph\appdata\local\programs\python\python36-32\lib\site-packages\click\core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\joseph\appdata\local\programs\python\python36-32\lib\site-packages\click\core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\joseph\appdata\local\programs\python\python36-32\lib\site-packages\click\core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "c:\users\joseph\appdata\local\programs\python\python36-32\lib\site-packages\ampy\cli.py", line 162, in ls
    for f in board_files.ls(directory, long_format=long_format):
  File "c:\users\joseph\appdata\local\programs\python\python36-32\lib\site-packages\ampy\files.py", line 124, in ls
    raise ex
  File "c:\users\joseph\appdata\local\programs\python\python36-32\lib\site-packages\ampy\files.py", line 117, in ls
    out = self._pyboard.exec_(textwrap.dedent(command))
  File "c:\users\joseph\appdata\local\programs\python\python36-32\lib\site-packages\ampy\pyboard.py", line 267, in exec_
    raise PyboardError('exception', ret, ret_err)
ampy.pyboard.PyboardError: ('exception', b'', b'Traceback (most recent call last):\r\n  File "<stdin>", line 6, in <module>\r\nOSError: [Errno 22] EINVAL\r\n')
`

MikeTeachman, 2018-08-31:
Same for Windows 7

joewez, 2018-09-02:
OK...

'''
C:\Users\Joseph>ampy -p com4 -b 115200 ls
boot.py
adc_test.py
app.py
aps_trusted.py
battlebot.py
bitmapfont.py
blink.py
bmp180_display.py
bmp180_test.py
button_async.py
button_dots.py
button_init.py
button_irq.py
button_test.py
captive_page.py
captive_server.py
carlos_clock.py
clock_demo.py
compass_test.py
config.py
custom_oled_test.py
dht_matrix.py
dht11_test.py
digit_clock.py
digit_test.py
dns_server.py
dnsquery.py
ds18b20_test.py
encoder_test.py
font5x8.bin
guess.py
gyro_test.py
hardware.py
hotspot_ftp.py
hotspot_web.py
initap.py
initwifi.py
lcd_test_1602.py
lcd_test_2004.py
life.py
lights.py
main.py
mastermind.py
matrix_bug.py
matrix_bug_blue.py
matrix_bug_red.py
matrix_demo.py
matrix_demo_blue.py
matrix_demo_red.py
matrix_particle.py
matrix_particle_red.py
matrix_read.py
matrix_read_blue.py
matrix_read_forever.py
matrix_scroll.py
matrix_scroll_blue.py
matrix_test.py
matrix_worm.py
matrix_worm_red.py
mqtt_test.py
npsoff.py
npstest.py
oled_demo.py
oled_demo2.py
oled_test.py
pcd_small.py
pcd_test.py
pixel.py
rgb.py
rtc.py
rtc_set.py
rtc_test.py
servo_demo.py
smalltext_test.py
smalltext_test2.py
starwars.py
strandtest.py
tft_test.py
tft_text.py
therm_test.py
thermo_demo.py
thermo_demo2.py
timers.py
tm1640_test.py
tones.py
uhttp.py
webrepl_cfg.py
widget.py
wifiboot.py
work_timer.py
wrapper.py
ws2812b_test.py
connect
dev
drivers
frozen
graphics
misc
profile
shields
umqtt
wwwroot
test.py
tft_dimensions.py
tft_random.py
tft_puzzle.py
tft_shield_test.py
tft_shield_test2.py
tft_shield_test3.py

C:\Users\Joseph>
'''

joewez, 2018-09-02:
Here is another try at a subdirectory...

'''
C:\Users\Joseph>ampy -p com4 -b 115200 ls wwwroot
Traceback (most recent call last):
  File "C:\Users\Joseph\AppData\Local\Programs\Python\Python37\Scripts\ampy-script.py", line 11, in <module>
    load_entry_point('adafruit-ampy==1.0.5', 'console_scripts', 'ampy')()
  File "c:\users\joseph\appdata\local\programs\python\python37\lib\site-packages\click\core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\joseph\appdata\local\programs\python\python37\lib\site-packages\click\core.py", line 697, in main
    rv = self.invoke(ctx)
  File "c:\users\joseph\appdata\local\programs\python\python37\lib\site-packages\click\core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\joseph\appdata\local\programs\python\python37\lib\site-packages\click\core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\joseph\appdata\local\programs\python\python37\lib\site-packages\click\core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "c:\users\joseph\appdata\local\programs\python\python37\lib\site-packages\ampy\cli.py", line 162, in ls
    for f in board_files.ls(directory, long_format=long_format):
  File "c:\users\joseph\appdata\local\programs\python\python37\lib\site-packages\ampy\files.py", line 124, in ls
    raise ex
  File "c:\users\joseph\appdata\local\programs\python\python37\lib\site-packages\ampy\files.py", line 117, in ls
    out = self._pyboard.exec_(textwrap.dedent(command))
  File "c:\users\joseph\appdata\local\programs\python\python37\lib\site-packages\ampy\pyboard.py", line 267, in exec_
    raise PyboardError('exception', ret, ret_err)
ampy.pyboard.PyboardError: ('exception', b'', b'Traceback (most recent call last):\r\n  File "<stdin>", line 6, in <module>\r\nOSError: [Errno 22] EINVAL\r\n')

C:\Users\Joseph>
'''

devxpy, 2018-09-02:
Have you tried prefixing the director by a slash?

Also, if you can, try this out using the master branch

```
$ git clone https://github.com/adafruit/ampy.git
$ cd ampy
$ pip install -e .
```

joewez, 2018-09-02:
Adding the slash prefix makes no difference with 1.0.5.

The good new is that the master branch seems to work...

'''
D:\Development\ampy>ampy -p com4 -b 115200 ls wwwroot
/wwwroot/afm.html
/wwwroot/applications.html
/wwwroot/editkeys.html
/wwwroot/getting.html
/wwwroot/images
/wwwroot/index.html
/wwwroot/reference.html
/wwwroot/replintro.html
/wwwroot/resources.html
/wwwroot/scripts
/wwwroot/styles
/wwwroot/usage.html

D:\Development\ampy>
'''

Although it does hang when I do add the slash....FYI.

closed 2019-01-21 by ladyada