scientifichackers/ampy #61: Possibility to rename files withing the board

yeyeto2788, 2018-08-08:
Hello guys! 

Just a quick possible  feature would be to add the option to `rename` files within the board, I don't really know if this is possible but I guess that it would be a kind of merge between:

*  `get` from the board and store it in memory
* `rm` from board.
* `put` the file stored in memory in the new directory.

Maybe something like this in the `cli.py`:
```python
@cli.command()
@click.argument("remote_file")
@click.argument("remote_new_file", type=click.File("wb"), required=True)
def rename(remote_file, remote_new_file):
    # Get the file contents.
    board_files = files.Files(_board)
    contents = board_files.get(remote_file)
    # Delete the provided file/directory on the board.
    board_files = files.Files(_board)
    board_files.rm(remote_file)
    # Put the file on the board.
    board_files.put(remote_filename, contents)
```

I know it involve more than just those couple of line but just wanted to help a bit.

Regards.

ladyada, 2019-01-21:
Hiya! We are discontinuing support for ampy, and will no longer be maintaining it. We are leaving this repository available for continued use. If you would like to take over supporting it, please contact us on the Adafruit Discord server and we can transfer the repository to you.
If you wish to continue developing it on your own, please fork the repository.

yeyeto2788, 2019-01-23:
Hello @devxpy, I guess I have it. When you work on a script, you run the script from an ampy command but then you want to rename the script into `boot.py` or `main.py` so you can now unplug the board and test if booting into the script works as expected.

This will also make our life easier since we do not need to type each command for doing it.

**e.g:**
First script `test.py` (basic script to output text on a LCD from an API) you run it on the board with `ampy -p <port> run test.py`.
Your first run went OK, so now want to rename the file on the board to `boot.py` and to do so you have to do the following:
* `ampy -p <port> rm test.py`
* Create a copy of the file on your computer and then call it `boot.py`
* `ampy -p <port> put boot.py`

I hope it makes sense what I just explained above 😄 

Regards.

devxpy, 2019-01-23:
I can work on adding an `mv` command, since it's the UNIX counterpart. Don't have a device with me right now, give me a few weeks.

yeyeto2788, 2019-01-27:
If you want me to test something just let me know!

Regards.

yeyeto2788, 2019-08-22:
Hello @devxpy, any updates here? Anything I can help with?

devxpy, 2019-08-22:
I'm devoting most of my time working on [ampy2](https://github.com/pycampers/ampy/tree/ampy2).

But if you have any code you want to upstream, I'll gladly accept your PR.

yeyeto2788, 2019-08-22:
And in ampy2 is the option? I can do the PR in the ampy2 instead of doing it in here.

devxpy, 2019-08-22:
ampy2 is not really usable right now. I'm still working on the [Dockerfile](https://github.com/micropython/micropython/pull/5003) for micropython.

yeyeto2788, 2019-08-22:
Really interesting, what I do now in order to not use a lot of memory is to [minify](https://github.com/yeyeto2788/MicroPythonScripts/blob/master/HelpScripts/PythonScripts/uncommend_and_upload.py) the script I'm loading into the board which might be cool to add on the ampy2.

The code I think it is documented and I could add the option on the new ampy2 if you wish. Of course, when you have a working version.

devxpy, 2019-08-22:
I'll try my best to include a plugin system to ampy2, so that features like this can be added independently. Something like [cargo's plugins](https://crates.io/categories/development-tools::cargo-plugins).

closed 2025-12-18 by yeyeto2788