scientifichackers/ampy #62: Does not work at Ubuntu 18.04 64

vitasam, 2018-08-08:
Hi,

can't get it working at Ubuntu 18.04 64 with python3
```
$ pip3 install adafruit-ampy
Collecting adafruit-ampy
Collecting pyserial (from adafruit-ampy)
. . .
Successfully installed adafruit-ampy-1.0.5 click-6.7 pyserial-3.4 python-dotenv-0.9.1
$ ampy --help

Command 'ampy' not found, did you mean:

  command 'mpy' from deb yorick-mpy-mpich2
  command 'mpy' from deb yorick-mpy-openmpi
  command 'empy' from deb python-empy
  command 'amph' from deb amphetamine

Try: sudo apt install <deb name>
```
The same result is for Python 2.x

dhalbert, 2018-08-08:
Do have `/home/<yourname>/.local/bin` in your `PATH`? I installed it, and that's where it put the executable.

vitasam, 2018-08-08:
thanks, works!
```
export PATH=$PATH:/home/$USER/.local/bin
$ ampy --version
ampy, version 1.0.5

```

closed 2018-08-08 by dhalbert

hoosierEE, 2018-09-19:
Adding `~/.local/bin` to my `PATH` solved this for me too (ubuntu 18.04).