scientifichackers/ampy #64: ampy not working while main.py is running

hyansuper, 2018-10-20:
If I have a main.py file in which a while-loop runs forever, then I can not issue the ampy command.
I don't think this is a bug, but I will have to use esptool.py to erase my esp32 borad, and flush firmware.
Is there any other way to edit or delete the none-stoping main.py file

jerryneedell, 2018-10-20:
if you can open a terminal session to the REPL  ( issue control C and enter to get to REPL) try:
```
import os
os.remove('main.py')

or
os.rename('main.py','was_main.py')
```

dhalbert, 2018-10-20:
@jerryneedell Do you think ampy might send a ctrl-C itself to the REPL, always, before doing something, so that any running program is interrupted? Is that a good enhancement?

jerryneedell, 2018-10-20:
@dhalbert I thought it did that already.

jerryneedell, 2018-10-20:
@dhalbert it should call enter_raw_repl()  https://github.com/adafruit/ampy/blob/master/ampy/pyboard.py#L175
before any operations. It's not clear why it won't break into main.py

dhalbert, 2018-10-20:
Yes, it does: https://github.com/adafruit/ampy/blob/master/ampy/pyboard.py#L180. Yes, we need more info.

jerryneedell, 2018-10-20:
This has come upon the past and been fixed then reappeared though I can't find the references to it.  Playing with the "delay" parameter might help ...

jerryneedell, 2018-10-20:
some discussion here: https://github.com/adafruit/circuitpython/issues/636

jerryneedell, 2018-10-20:
FWIW -- I am able to do am ampy ls with an endless main.py on my esp32.

```

while True:
    print("looping in main.py")

jerryneedell@Ubuntu-Macmini:~/projects/esp32$ ampy ls
/aio_publish.py
/main.py
```
It did hang once but it recovered when I tried adding a longer delay 
ampy -d 1.5

I usuallh have the delay set to 0  -- here is my .ampy file
```
jerryneedell@Ubuntu-Macmini:~/projects/esp32$ cat .ampy
# Example .ampy file
# Please fill in your own port, baud rate, and delay
AMPY_PORT=/dev/ttyUSB0
AMPY_BAUD=115200
AMPY_DELAY=0

```
I am using ampy 1.0.5  -- Ubuntu 18.05  - micropython 1.9.4
MicroPython v1.9.4-631-g338635ccc on 2018-10-10; ESP32 module with ESP32

ladyada, 2019-01-21:
Hiya! We are discontinuing support for ampy, and will no longer be maintaining it. We are leaving this repository available for continued use. If you would like to take over supporting it, please contact us on the Adafruit Discord server and we can transfer the repository to you.
If you wish to continue developing it on your own, please fork the repository.

willcharlton, 2019-02-21:
Does anyone know of a good replacement for `ampy` since it is no longer being supported?

jerryneedell, 2019-02-22:
amp is alive and well -- it is now maintained at https://github.com/pycampers/ampy

devxpy, 2019-02-27:
@willcharlton Hey! I'm one of the members of [pycampers](https://github.com/pycampers); we just got maintainership of this project.

If you have any specific stuff you want to talk about, just ping me.

imranparuk, 2019-04-27:
Still having this issue...

imranparuk, 2019-04-27:
Still having this issue, 

I have tried @gkyle approach, however no luck. In fact it stops me from being able to execute any commands.

devxpy, 2019-04-28:
I don't have a clear solution, but I just wanna spit out my 2 cents :-

This project is originally based on `pyboard.py`, which to me, looks like a hack. It _tries_ to control the device, by essentially, sending keyboard shortcuts, and waiting for absurd things like a `>` character to appear on the output. When it doesn't, (due to some connectivity issues, or keyboard events not being sent at the "right" time, or whatever), it panics, and we get issues like this, and many others. So it's essentially automating stuff that you would otherwise do inside a `screen` session.

What I would like to see, is a _proper_ API for communicating with the device. 

I admit, that haven't had the time to fully devote myself to this project, but this is one fundamental issue I see with micropython, in the limited time I have the chance to fire up my editor, and hack on ampy.

I would love any suggestions that you guys have on building a proper communication backend inside micropython. 

I have one idea -- Maybe we can build a special, "debug" firmware, that contains a simple http/tcp server, that listens for commands from an ampy client. This way, we could avoid all sorts of issues that arise from doing flaky serial communication, and potentially,  even enable in-browser apps that talk directly to the boards!

dhalbert, 2019-04-28:
@devxpy `ampy`, [`rshell`](https://github.com/dhylands/rshell), and similar programs are designed to compensate for the lack of some other kind of interface, such as mass storage (MSC) access to the on-board filesystem. You might look at `rshell` to see if it works somewhat better, and adapt `ampy` to use its techniques, or simply stop supporting `ampy`.

As you see, trying to control things through the REPL can be less than satisfactory. That's one reason why Adafruit dropped support for `ampy`, and is also not supporting boards without MSC for CircuitPython 4.0 and up.

devxpy, 2019-10-08:
@dhalbert Took me time to get there, but I've finally built a POC, that involves a development micropython firmware, that hosts an RPC server. This allows users to remotely execute code on the device, from their machine, without ever interacting through the REPL in serial mode.

More details [here](https://github.com/scientifichackers/ampy/blob/ampy2/README.md).

Needless to say, haven't quite given up on ampy just yet :)

pratik-pato, 2022-04-28:
Adding a delay of 0.5 worked for me.
Try
`ampy -p <port> -d 0.5 <rest of the command>`

willcharlton, 2022-04-28:
@devxpy - whoa, that is great news. I'm just seeing this now that you're the new maintainer. I will start thinking about features again. Thank you!

N1ckn1ght, 2022-05-17:
> Adding a delay of 0.5 worked for me. Try `ampy -p <port> -d 0.5 <rest of the command>`

This is working, except that for me it was more than 0.5, so for other people I suggest experimenting.