scientifichackers/ampy #65: Unable To list subdirectories

yurikleb, 2018-10-25:
I'm unable to list subdirectories on an ESP32.

I've successfully created a subdirectory under /flash/
and copied a file into it using the following commands:
`
export AMPY_PORT=/dev/ttyUSB0
ampy mkdir /flash/new_dir
ampy put myscript.py /flash/new_dir/myscript.py
`

then when running:
`ampy ls /flash/new_dir/`
I'm getting the following error:

`
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/ampy/files.py", line 117, in ls
    out = self._pyboard.exec_(textwrap.dedent(command))
  File "/usr/local/lib/python3.6/dist-packages/ampy/pyboard.py", line 267, in exec_
    raise PyboardError('exception', ret, ret_err)
ampy.pyboard.PyboardError: ('exception', b'', b'Traceback (most recent call last):\r\n  File "<stdin>", line 6, in <module>\r\nOSError: [Errno 2] ENOENT\r\n')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/ampy", line 11, in <module>
    load_entry_point('adafruit-ampy==1.0.5', 'console_scripts', 'ampy')()
  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/ampy/cli.py", line 162, in ls
    for f in board_files.ls(directory, long_format=long_format):
  File "/usr/local/lib/python3.6/dist-packages/ampy/files.py", line 122, in ls
    raise RuntimeError('No such directory: {0}'.format(directory))
RuntimeError: No such directory: /flash/new_dir/
`

jerryneedell, 2018-10-25:
try 
```
amp ls -r
```
it should list all files in all subdirectories. Not exactly what you want, but it will get you the information...

yurikleb, 2018-10-25:
Hey, thanks for the quick response,
I'm getting:
`Error: no such option: -r`

jerryneedell, 2018-10-25:
just to check, I think the -r must go at the end : ampy ls -r

yurikleb, 2018-10-25:
no luck :(
here is a screenshot:
[image]

jerryneedell, 2018-10-25:
Sorry -- I'll check it on my hardware later this afternoon (EDT).

yurikleb, 2018-10-25:
Sure no rush!
also checked
`ampy ls --help`
the only available option seems to be `-l, --long_format`
which prints the file sizes

yurikleb, 2018-10-25:
For now, I ended up using the `os.listdir('my_dir')` command via the REPL.
But would be great to be able to browse subfolders via ampy :)

dhalbert, 2018-10-25:
Looking at the code history, `-r` was added since 1.0.5 was released, and the version number wasn't bumped yet. So the latest release (1.05) doesn't include it. We need to make a new release.

yurikleb, 2018-10-25:
looking forward!

dhalbert, 2018-10-25:
Yeah, this is confusing. The 1.0.5 version number bump was done before `-r` was added. So there's a github source release that's up to date, but the pypi.org release is July 9, before the additional merges: https://pypi.org/project/adafruit-ampy/#history

ampy really needs to be travis-ified - then the pypi releases would happen automatically.

jerryneedell, 2018-10-25:
once the updated version gets released -- here is what it looks like:
```
jerryneedell@Ubuntu-Macmini:~/projects/esp32$ ampy ls -r
/boot.py
/lib/bitmapfont.mpy
jerryneedell@Ubuntu-Macmini:~/projects/esp32$ ampy ls --help
Usage: ampy ls [OPTIONS] [DIRECTORY]

  List contents of a directory on the board.

  Can pass an optional argument which is the path to the directory.  The
  default is to list the contents of the root, /, path.

  For example to list the contents of the root run:

    ampy --port /board/serial/port ls

  Or to list the contents of the /foo/bar directory on the board run:

    ampy --port /board/serial/port ls /foo/bar

  Add the -l or --long_format flag to print the size of files (however note
  MicroPython does not calculate the size of folders and will show 0 bytes):

    ampy --port /board/serial/port ls -l /foo/bar

Options:
  -l, --long_format  Print long format info including size of files.  Note the
                     size of directories is not supported and will show 0
                     values.
  -r, --recursive    recursively list all files and (empty) directories.
  --help             Show this message and exit.

```

dhalbert, 2018-10-26:
I made a 1.0.7 release (1.0.6 was skipped due to tagging issues). Could you use `pip3` to upgrade to that release and try again?

jerryneedell, 2018-10-26:
1.0.7 works OK for me.on  ESP32 (Mocropython 1.94)  and feather_nrf52832 (CircuitPython 4.4.0 alpha2)

yurikleb, 2018-10-27:
1.0.7
Works great! 
Both the -r option and running `ls` on a subdirectory doe example:
`ampy ls /flash/some_sub_dir/`
Thanks very much for the quick fix, much appreciated!

closed 2018-10-27 by yurikleb