scientifichackers/ampy #66: can't establish a connection anylonger after upload of main.py

nichtleiter, 2018-10-26:
On my system runs: `ampy --version` → `ampy, version 1.0.7`
Used board: wemos lolin d32

After i uploaded a simple blinking example `ampy` refuses to establish a connection to my board.

Invoking `ampy -p /dev/ttyUSB0 ls` yields:

```
ampy -p /dev/ttyUSB0 ls
Traceback (most recent call last):
  File "/home/johannes/.local/lib/python3.6/site-packages/ampy/files.py", line 164, in ls
    out = self._pyboard.exec_(textwrap.dedent(command))
  File "/home/johannes/.local/lib/python3.6/site-packages/ampy/pyboard.py", line 265, in exec_
    ret, ret_err = self.exec_raw(command)
  File "/home/johannes/.local/lib/python3.6/site-packages/ampy/pyboard.py", line 256, in exec_raw
    self.exec_raw_no_follow(command);
  File "/home/johannes/.local/lib/python3.6/site-packages/ampy/pyboard.py", line 253, in exec_raw_no_follow
    raise PyboardError('could not exec command')
ampy.pyboard.PyboardError: could not exec command

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/johannes/.local/bin/ampy", line 11, in <module>
    sys.exit(cli())
  File "/home/johannes/.local/lib/python3.6/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/johannes/.local/lib/python3.6/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/johannes/.local/lib/python3.6/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/johannes/.local/lib/python3.6/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/johannes/.local/lib/python3.6/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/johannes/.local/lib/python3.6/site-packages/ampy/cli.py", line 194, in ls
    for f in board_files.ls(directory, long_format=long_format, recursive=recursive):
  File "/home/johannes/.local/lib/python3.6/site-packages/ampy/files.py", line 168, in ls
    if ex.args[2].decode("utf-8").find("OSError: [Errno 2] ENOENT") != -1:
IndexError: tuple index out of range
```

`rshell` exists with a similar error (see https://github.com/dhylands/rshell/issues/27#issuecomment-433260583).

At the same time `mpfshell` works without complaining. Here a quick sample session i started right after invoking `ampy`:

```
johannes:~/iot/esp/Micropython/MyExamples/TimerTest % mpfshell -c "open ttyUSB0"
Connected to esp32

** Micropython File Shell v0.8.1, sw@kaltpost.de **
-- Running on Python 2.7 using PySerial 3.4 --

mpfs [/]> cat boot.py
# This file is executed on every boot (including wake-boot from deepsleep)
#import esp
#esp.osdebug(None)
#import webrepl
#webrepl.start()

mpfs [/]> cat main.py
import machine
from machine import Pin
from machine import Timer
import micropython

micropython.alloc_emergency_exception_buf(256)

print("alloc_emergency_exception_buf done")

io23 = Pin(23, Pin.OUT)
io05 = Pin(5, Pin.OPEN_DRAIN)

io23.value(1)
io05.value(0)

timer1 = Timer(-1)
timer1.init(mode=Timer.PERIODIC, period=100, callback=lambda t:io23.value(not io23.value()))

timer2 = Timer(-2)
timer2.init(mode=Timer.PERIODIC, period=100, callback=lambda t:io05.value(not io05.value()))

print("Let's start blinking")

mpfs [/]>
```

dhalbert, 2018-10-26:
That's an unusual main.py, because it exits instead of looping forever. Do you have the same problem with a `while True: pass` type of main.py?

Given that there are two shell programs that don't work, and one that does, it would be worth looking at what's actually being sent. Also, try a `--delay` option of a few seconds in `ampy` and see if that helps.

nichtleiter, 2018-11-01:
Finally i found some time to test your case.

If i `touch` a new `main.py` with the content
```
while True: pass
```
everything works as expected.

However if i put
```
import machine
from machine import Pin
from machine import Timer
import micropython

micropython.alloc_emergency_exception_buf(256)

print("alloc_emergency_exception_buf done")

io23 = Pin(23, Pin.OUT)
io05 = Pin(5, Pin.OPEN_DRAIN)

io23.value(1)
io05.value(0)

timer1 = Timer(-1)
timer1.init(mode=Timer.PERIODIC, period=10, callback=lambda t:io23.value(not io23.value()))

timer2 = Timer(-2)
timer2.init(mode=Timer.PERIODIC, period=500, callback=lambda t:io05.value(not io05.value()))

print("Let's start blinking")

while True: pass
```
in my `main.py` (my original file), `ampy` works very unreliable. If i execute `ampy -p /dev/ttyUSB0 put main.py` successively, ampy fails 2 times out of 3 with the following error message
```
Traceback (most recent call last):
  File "/home/johannes/.local/bin/ampy", line 11, in <module>
    sys.exit(cli())
  File "/home/johannes/.local/lib/python3.6/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/johannes/.local/lib/python3.6/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/johannes/.local/lib/python3.6/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/johannes/.local/lib/python3.6/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/johannes/.local/lib/python3.6/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/johannes/.local/lib/python3.6/site-packages/ampy/cli.py", line 263, in put
    board_files.put(remote, infile.read())
  File "/home/johannes/.local/lib/python3.6/site-packages/ampy/files.py", line 209, in put
    self._pyboard.exec_("f = open('{0}', 'wb')".format(filename))
  File "/home/johannes/.local/lib/python3.6/site-packages/ampy/pyboard.py", line 265, in exec_
    ret, ret_err = self.exec_raw(command)
  File "/home/johannes/.local/lib/python3.6/site-packages/ampy/pyboard.py", line 256, in exec_raw
    self.exec_raw_no_follow(command);
  File "/home/johannes/.local/lib/python3.6/site-packages/ampy/pyboard.py", line 253, in exec_raw_no_follow
    raise PyboardError('could not exec command')
ampy.pyboard.PyboardError: could not exec command
```

Again, if i put `main.py` with `mpfshell -n -c "open ttyUSB1; put main.py"` on the board, everything works.

As a next step i will try to compare the code of `mpfshell` with `ampy`. I hope i find some time for this soon.

rrottmann, 2019-01-14:
@nichtleiter: Did you check the delay option as suggested?  Got the same error message after adding wifi connection to the code. Setting the delay fixed ampy in my case.

> -d, --delay DELAY  Delay in seconds before entering RAW MODE (default 0).
>                     Can optionally specify with AMPY_DELAY environment
>                     variable.