scientifichackers/ampy #67: Replace string-based functions with actual functions

j4k0bk, 2018-11-09:
Defining functions inside strings (to run on device) is messy, and the source is not testable. Another way to define a function on device via raw_repl is to define the function locally and then do:

pyb.exec(textwrap.dedent(inspect.getsource(func)))

For instance in ampy/files.py line 109 "listdir" is defined on device. Another way would be to define this method normally inside files.py and then:

pyb.exec(textwrap.dedent(inspect.getsource(listdir)))

ladyada, 2019-01-21:
Hiya! We are discontinuing support for ampy, and will no longer be maintaining it. We are leaving this repository available for continued use. If you would like to take over supporting it, please contact us on the Adafruit Discord server and we can transfer the repository to you.
If you wish to continue developing it on your own, please fork the repository.

devxpy, 2019-01-22:
I would probably have code inside strings than do `inspect.getsource()`. You cannot test code that is meant to run on micropython anyway.

That said, If you have a better way of doing it, like actual `.py` files with some kind of tool that's built specifically for code generation, like [Cog](https://nedbatchelder.com/code/cog/), I would gladly accept a pull request.

devxpy, 2019-02-22:
I think this can be replaced with an RPC like approach, where a daemon runs on the micro python side, and communicates using either TCP or serial.

This can also potentially solve serial communication-related bugs, and circumvent the multiple serial connections thing.

curiouswala, 2021-02-08:
Agreed. We are working on a rewrite of ampy which uses WebREPL instead of USB. Will make an ALPHA release soon.

closed 2021-02-08 by curiouswala