scientifichackers/ampy #70: Error while trying to push file

lazarvgd, 2018-11-30:
Hi there,

Recently I have found that ampy has unknown issues for me. When I try to push file to ESP8266 board, from time to time I get error:

`[svarog@hal9000 ~]$ sudo ampy --port /dev/ttyUSB0 put lcd.py main.py
Traceback (most recent call last):
  File "/usr/bin/ampy", line 10, in <module>
    sys.exit(cli())
  File "/usr/lib/python3.7/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3.7/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/usr/lib/python3.7/site-packages/ampy/cli.py", line 263, in put
    board_files.put(remote, infile.read())
  File "/usr/lib/python3.7/site-packages/ampy/files.py", line 218, in put
    self._pyboard.exec_("f.write({0})".format(chunk))
  File "/usr/lib/python3.7/site-packages/ampy/pyboard.py", line 265, in exec_
    ret, ret_err = self.exec_raw(command)
  File "/usr/lib/python3.7/site-packages/ampy/pyboard.py", line 256, in exec_raw
    self.exec_raw_no_follow(command);
  File "/usr/lib/python3.7/site-packages/ampy/pyboard.py", line 253, in exec_raw_no_follow
    raise PyboardError('could not exec command')
ampy.pyboard.PyboardError: could not exec command
`

If I try another time it will succeed.
But there are, let's say edge cases, when I push few files to the board, for some reason what ever I do I am getting the same error. In order to avoid this error, I need to flash whole board, write firmware again and push files.

Please let me know if you need any additional info or if this is already known issue(I searched for it but coudn't find the thread)

thanks,
Lazar :)

cameronbunce, 2019-01-21:
I'm not sure if I'm seeing the same thing, but I've updated ampy 'pip3 install --upgrade ampy' and I'm seeing similar errors:

` ampy -p /dev/ttyUSB0 put adafruit_ht16k33/
Traceback (most recent call last):
  File "/usr/local/bin/ampy", line 11, in <module>
    load_entry_point('adafruit-ampy==1.0.7', 'console_scripts', 'ampy')()
  File "/home/pi/.local/lib/python3.5/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/.local/lib/python3.5/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/pi/.local/lib/python3.5/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/pi/.local/lib/python3.5/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/.local/lib/python3.5/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/.local/lib/python3.5/site-packages/ampy/cli.py", line 248, in put
    board_files.mkdir(remote_parent)
  File "/home/pi/.local/lib/python3.5/site-packages/ampy/files.py", line 201, in mkdir
    raise ex
  File "/home/pi/.local/lib/python3.5/site-packages/ampy/files.py", line 192, in mkdir
    out = self._pyboard.exec_(textwrap.dedent(command))
  File "/home/pi/.local/lib/python3.5/site-packages/ampy/pyboard.py", line 267, in exec_
    raise PyboardError('exception', ret, ret_err)
ampy.pyboard.PyboardError: ('exception', b'', b'Traceback (pinakahuling huling tawag): \r\r\n  File "<stdin>", line 6, sa <module>\r\r\nOSError: [Errno 17] Mayroong file\r\n')
`

ladyada, 2019-01-21:
Hiya! We are discontinuing support for ampy, and will no longer be maintaining it. We are leaving this repository available for continued use. If you would like to take over supporting it, please contact us on the Adafruit Discord server and we can transfer the repository to you.
If you wish to continue developing it on your own, please fork the repository.

cameronbunce, 2019-01-22:
I was flipping between a few ways that I got errors and determined that the latest CircuitPython release for the 8266 was the common point. Ampy was able to Read and write to the MicroPython releases on the 8266 and the errors I was getting contained text from the esp side of things more than Ampy. 
Try the latest MicroPython releases if you’re having trouble with the 8266

devxpy, 2019-01-23:
@lazarvgd  Cam you confirm whether Micropython works for you?

P.S. Circuitpython has dropped support for ESP8266.

closed 2020-01-11 by infogeek