scientifichackers/ampy #76: Is it okay to drop legacy Python support?

devxpy, 2019-02-25:
I was wondering if anyone, in particular, has an issue with me dropping support for anything older than 3.6

Considering that this is a CLI tool, I fail to see the significance of maintaining backwards compat. 

In the future, I will try to release binaries using something like [PyOxidizer](https://github.com/indygreg/PyOxidizer), which would even eliminate the need to have python installed.

ZairMahmood, 2019-02-27:
What features are being implemented from Python 3.6 that would require drop for backwards compatibility?

devxpy, 2019-02-27:
@ZairMahmood Nothing in particular. I never said that we _require_ to drop support.

It's just a matter of development speed, I guess. If we didn't have to worry about supporting older pythons, I feel like we can work faster on the project, overall.

For starters, I just upgraded `ampy run` to stream stdout from scripts, so we don't have to wait for the script to finish, before seeing the output.

Although I would love to hear an argument _against_ doing this, If you have one.

ZairMahmood, 2019-02-27:
My argument would be that someone implementing ampy directly into their python application via import rather than just using the CLI, they are not limited to using only Python 3.6.

infogeek, 2020-01-11:
I don't see any strong arguments to support legacy python anymore. It will make more sense to do it in ampy2 release and archive the current version.

cefn, 2020-04-19:
It's not an argument against dropping support, but for reference I would find my project [vgkits-vanguard](https://pypi.org/project/vgkits-vanguard/) had broken which so far had a dependency of...

```
"adafruit-ampy>=1.0.3",
```
https://github.com/vgkits/vanguard-cli/blob/master/setup.py#L31

...and I believe even runs in python2 right now (it's been a while since I tried). Certainly an aim of the project is to be able to run on various people's python build, whatever they have.

Probably I should have pinned that to a major version anyway, and if I had practiced good hygiene when building it, that would mean dropping support as part of a major version bump would do no harm.

TonyLHansen, 2020-05-18:
I've been gradually adding this statement to various python project of mine:

import sys
assert sys.version_info >= (3, 6)

I definitely see no need to support python2 any more.

closed 2021-02-08 by curiouswala