scientifichackers/ampy #90: Cannot upload file with specific starting character on Windows

lubosmato, 2020-01-27:
## How to reproduce (only Windows)
Upload a file using `Files.put`. File must be inside a folder and a filename must start with a character from set: `a`, `b`, `f`, `n`, `r`, `t`, `v`, `o`, `x`. For example imagine this filepath `folder\file_starting_with_f.py`. Windows uses `\`. The problem occurs on line https://github.com/scientifichackers/ampy/blob/master/ampy/files.py#L214 

```python
# filename == "folder\file_starting_with_f.py"
# '\f' is escape character so opening a file with escape character will fail with EACCES
self._pyboard.exec_("f = open('{0}', 'wb')".format(filename))
```

Probably same issue which breaks tests on Windows #72

closed 2020-01-27 by lubosmato

devxpy, 2020-01-28:
Was this solved?

lubosmato, 2020-01-28:
No, it wasn't because it is not an issue in ampy. I thought it is related to ampy project but the issue is related to intellij-micropython which uses ampy.