scientifichackers/ampy #96: ampy.pyboard.PyboardError: could not enter raw repl

GabrieleAnsaldo, 2020-12-06:
Hello,

I am trying to run a python script on my ESP32 however I get this error: 'could not enter raw repl'

```
user@server ~ % ampy --port /dev/tty.usbserial-0001 --baud 115200 ls
b'MicroPython v1.13-221-gc8b055717 on 2020-12-06; ESP32 module with ESP32\r\nType "help()" for more information.\r\n>>> '
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.8/bin/ampy", line 8, in <module>
    sys.exit(cli())
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/ampy/cli.py", line 194, in ls
    for f in board_files.ls(directory, long_format=long_format, recursive=recursive):
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/ampy/files.py", line 167, in ls
    self._pyboard.enter_raw_repl()
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/ampy/pyboard.py", line 196, in enter_raw_repl
    raise PyboardError('could not enter raw repl')
ampy.pyboard.PyboardError: could not enter raw repl
```

What is causing it? If I connect to my eps32 with screen (on mac) directly on the serial it works.

thinkJD, 2021-01-18:
I have the same problem. 
My boot.py and main.py is empty. Maybe this has something to do with it. WIll investigate further.

boxed, 2021-01-27:
Presumably this is the same as https://github.com/scientifichackers/ampy/issues/19 

I had this same problem and passing `-d 0.5` indeed made it possible for me to execute the ls command.

curiouswala, 2021-02-08:
As @boxed  mentioned this is a common issue. Try adding a delay to the AMPY command, it seems to work in most cases.
Some boards just won't work reliably in my case so try a different board and USB cable.

closed 2021-02-08 by curiouswala

ciscojmg, 2022-05-19:
Tengo este mismo problema. Tengo que estar conectando y desconectando el FTDI para que pueda funcionar con "normalidad".

ciscojmg, 2022-05-19:
Conectense por serial, dejalo en REPL y desconectate del puerto serial.
Sugerencia de un compannero.