scientifichackers/ampy #116: timeout waiting for first EOF reception

MaxKru, 2022-08-09:
hi,
i try to switch from thonny to vsc, i managed to get a program working in vsc
which is:

"""
from machine import Pin
from time import sleep

led1 = Pin(17,Pin.OUT,value = 0)

btn = Pin(16,Pin.IN,Pin.PULL_UP)

print(led1)

while True:
    if btn.value() == 1:
        led1.value(1)
    else:
        led1.value(0)
"""

the problem im running into is: after pressing the button btn, the programm loop is cancelt
and the programm is closed

the command im using is, ampy --port COM3 run  blinkmicropython.py
i also tried [..] run -n

i would like to keep this programm running forever, and not cancel it after one runthrough.
how can i accomplish this?

YeesterPlus, 2024-07-14:
aggree

YeesterPlus, 2024-07-14:
use ` -n ` , but it won't print