scientifichackers/ampy #125: CAUTION - THIS PROJECT IS STALE (UNMAINTAINED).  There is no point submitting PRs into it - nobody merges any of them...

gitcnd, 2024-01-16:
Pity, since it's a nice tool, and since lots of us tried to contribute, but all our efforts just drop on the floor...

dhalbert, 2024-01-16:
Anyone is welcome to fork this and pick up maintainenace. Adafruit gave up maintenance and scientifichackers took over. Could someone else do that now?

gitcnd, 2024-01-16:
I'm happy to look after it, and have a massive selection of boards and PC/Mac/linux hosts for testing.  If you know how to add me to allow this, please do (I'm kinda new to contributing through GitHub to other peoples work, so if I'm supposed to do something from my side - let me know.).  More about me is here: https://www.linkedin.com/in/drakechristopher/

gitcnd, 2024-01-16:
I'll ask ( https://github.com/scientifichackers/ampy/pull/126 )

vvvictor07, 2024-02-15:
Hi @gitcnd any news around this?
I am also interested to make this project work again (because this [one](https://github.com/curiouswala/tangiblepy) is not an replacement) and I also interested to update its [package](https://github.com/JetBrains/intellij-micropython), because there are some projects depending on it, e.g. [intellij-micropython](https://github.com/JetBrains/intellij-micropython)

If there will be no news, i will fork this repo, review and merge exist PRs, contact [adafruit-python](https://pypi.org/user/adafruit-python/) to update package, because it seems that project got stucked.

dhalbert, 2024-02-15:
We have talked a little with the MicroPython folks about making `mpremote`, their preferred CLI tool, compatible with CircuitPython. Would `mpremote` be an acceptable substitute? We don't feel we need such a tool for CircuitPython, but that doesn't mean other people wouldn't want one.

gitcnd, 2024-02-16:
ampy uses pyboard "under the hood".

mpremote is a rewrite of pyboard (exact same source code, all the same bugs)

the source code is a very basic and generally unintelligent "hack" that's hardcoded to look for specific responses (with bugs in their timeout code) and generally has zero robustness.

I fixed and patched most of that and submitted PRs, but there's little chance those will get incorporated anytime soon (one is a bug with python itself, but it has a workaround).  micropython is infamous for rarely/never adopting PRs (or taking years) and generally being hostile to outside help.

I recommend to switch to circuitpython - the micropython fork is too unprofessional and too far behind to rely on.

jdimpson, 2024-03-01:
@dhalbert I've noticed that Adafruit has started building CircuitPython images for ESP32C3 chips (which don't support USB drive interface). Surely Adafruit uses something similar to ampy to automate their testing. Do you happen to know what it is?

dhalbert, 2024-03-01:
We suggest Thonny, and we have a web workflow:
https://learn.adafruit.com/getting-started-with-web-workflow-using-the-code-editor
https://learn.adafruit.com/circuitpython-with-esp32-quick-start

gitcnd, 2024-05-09:
Thonny and the adafrruit online workflow both suffer the RTS/DTR pin setting problems which make them "not work" on boards which wire those pins to Reset and GPIO0 :-(

I just modified my local ampy - it works great with circuitpython now.  All that's needed this this in ampy/files.py

```
            try:
                import ubinascii
            except:
                import binascii as ubinascii
```

gitcnd, 2024-05-09:
My new code is here:  https://github.com/gitcnd/ampy